MagicMirrorOrg/MagicMirror #1492: Creating an image of a clean installation of MagicMirror

ishepard, 2018-12-29:
I see a lot of people having issues on installing dependencies, MagicMirror, etc. 
Maybe it could be useful to create an image that people can use to format their SD card (i.e., using `dd`) containing the basic installation of MagicMirror. 
We can create an image for Raspberry Pi 3, 3+, of 8 and 16GB. So 4 images in total.

What do you think?

MichMich, 2018-12-29:
It will be a lot of work to maintain, and defeats the purpose of the Project. (https://github.com/MichMich/MagicMirror#manifesto)

But feel free to create and host a image yourself! :)

ShrinivasPuranik, 2018-12-29:
There is a repo that does that maintained by skarbat [https://github.com/skarbat/raspimirror](https://github.com/skarbat/raspimirror). Though I dont think there is an automated build.

MichMich, 2018-12-30:
This solves this issue. Closing it for now. Feel free to let me know if I need to reopen.

closed 2018-12-30 by MichMich