MagicMirrorOrg/MagicMirror #1493: MagicMirror Forum

Batminer, 2018-12-29:
Total noob. Been working on a mirror for a few weeks. 
I am having an issue using the forum, https://forum.magicmirror.builders/
1) When I am trying to post in the troubleshooting section I get the following pop up;
Your email has not been confirmed yet, please click here to confirm your email.
I have been clicking the link in the pop up and the button on the edit profile page half a dozen times with three different emails (@outlook, @yahoo & @gmail)
I don't see anything anywhere in any of my email inboxes. 
What gives
Merry New Year!
Sincerely,
Andy Wa

MichMich, 2018-12-30:
@paviro Can you help?

MichMich, 2019-01-17:
@paviro @Batminer is this issue solved?

Batminer, 2019-03-19:
Yes it looks like this has been corrected.
Thank you

closed 2019-04-17 by MichMich