MagicMirrorOrg/MagicMirror #1499: new weather module

apm1007, 2019-01-02:
i found three problems with the new weather module
decimal signal isnt working. 
[image]
[image]
layout of forecast ist far to big. dunno why there are empty lines compared to the old forecast
[image]
hour forecast isnt possible.
Pretty Good idea for taking indoor temp and humidity to the weather module! but would be great to get the symbols vertical for better screen use.
[image]
Thank you!

MichMich, 2019-01-02:
@apm1007  Thanks for your feedback! 

@fewieden do you want me to tag you in these issue reports?

fewieden, 2019-01-03:
@MichMich sure you can tag me it will not hurt, but I'm back at work now, so I can't promise anything.

1. Decimal symbol was not implemented, but still documented in the readme. I added it and it will be in the next pr.

2. I'm not sure yet why the forecast table has this spacing. Will have a look when I find time to debug this.

3. The indoor values are horizontal like in the currentweather module. I made them now smaller and vertical. Also added a class to it so people can style it individually. Will be in next pr.

fewieden, 2019-01-05:
All 3 points are handled in https://github.com/MichMich/MagicMirror/pull/1510

MichMich, 2019-01-08:
Merged.

closed 2019-01-08 by MichMich