MagicMirrorOrg/MagicMirror #1500: Electron v2 Fails to Open with OpenGL Driver Enabled (Nov 2018 Raspbian)

shbatm, 2019-01-02:
Electron fails to launch when OpenGL (Full KMS or Fake KMS) is enabled.

Upgrading to Electron v3.0.13 seems to fix the issue, but don't know if this breaks something else.  Cannot use Electron v4 yet with Raspberry Pi due to dependency issue (electron/electron#16205)

**Platform**: 
Electron v2.0.4 and v2.0.16
Raspberry Pi 3b
Raspbian with Desktop - November 2018 image

**Node Version**: v8.15.0

**MagicMirror Version**: V2.6.0 and V2.7.0-dev

**Description**: Provide a detailed description about the issue and include specific details to help us understand the problem. Adding screenshots will help describing the problem.

**Steps to Reproduce**: 
1. Perform fresh install of MagicMirror on clean version of Raspbian w/ Desktop Nov-2018 image
2. Enable OpenGL driver via `sudo rasbpi-config`
3. Restart
4. Attempt to start MagicMirror with `npm start`

**Expected Results**: 
MagicMirror application opens

**Actual Results**: Describe what you actually saw.
Hangs at "Launching application". No error messages.

**Configuration**:
Default.

Skorpionbird, 2019-01-02:
Same issue after i have installed the update.
Actual Results: Describe what you actually saw.
Hangs at "Launching application". 
with this error message:

pi@raspberrypi:~/MagicMirror $ DISPLAY=:0 npm start

> magicmirror@2.6.0 start /home/pi/MagicMirror
> sh run-start.sh

Starting MagicMirror: v2.6.0
Loading config ...
Loading module helpers ...
No helper found for module: alert.
Initializing new module helper ...
Module helper loaded: updatenotification
No helper found for module: clock.
Initializing new module helper ...
Module helper loaded: calendar
No helper found for module: compliments.
Initializing new module helper ...
Module helper loaded: MMM-WatchDog
Initializing new module helper ...
Module helper loaded: MMM-Remote-Control
No helper found for module: MMM-SystemStats.
No helper found for module: MMM-Dynamic-Modules.
No helper found for module: MMM-pages.
No helper found for module: MMM-page-indicator.
No helper found for module: MMM-Hover.
No helper found for module: MMM-NetworkScanner.
No helper found for module: MMM-Cursor.
No helper found for module: currentweather.
No helper found for module: weatherforecast.
Initializing new module helper ...
Module helper loaded: newsfeed
No helper found for module: MMM-SwissStationboard.
No helper found for module: MMM-SwissCommute.
No helper found for module: MMM-BackgroundSlideshow.
No helper found for module: octomirror-module.
Initializing new module helper ...
Module helper loaded: MMM-GoogleMapsTraffic
Initializing new module helper ...
Module helper loaded: MMM-PIR-Sensor
All module helpers loaded.
Starting server on port 8080 ...
Server started ...
Connecting socket for: updatenotification
Connecting socket for: calendar
Starting node helper for: calendar
Connecting socket for: MMM-WatchDog
Starting module helper: MMM-WatchDog
Connecting socket for: MMM-Remote-Control
Starting node helper for: MMM-Remote-Control
Connecting socket for: newsfeed
Starting module: newsfeed
Connecting socket for: MMM-GoogleMapsTraffic
Starting node_helper for module: MMM-GoogleMapsTraffic
Connecting socket for: MMM-PIR-Sensor
Sockets connected & modules started ...
Launching application.
Wed Jan 02 2019 23:14:18 GMT+0100 (CET) - WatchDog: Heartbeat timeout. Frontend                                                                                         might have crashed. Exit now.
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! magicmirror@2.6.0 start: `sh run-start.sh`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the magicmirror@2.6.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional log                                                                                        ging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2019-01-02T22_14_19_205Z-debug.log

shbatm, 2019-01-03:
@Skorpionbird -- Out of curiosity, try updating electron and see if it fixes this issue/breaks something else:

```shell
cd ~/MagicMirror
sed -i 's/electron": "\^2\.0\..*",/electron": "^3.0.13",/g' package.json
npm install
```

MichMich, 2019-01-03:
@shbatm You don't need to modify the package.json file. Simply use the following command:

```npm install electron@3.0.13```

Let me know if this fixes the issue.

MichMich, 2019-01-03:
I've bumped the Electron dependency to 3.0.13 in the develop branch. Please let me know if that fixes the issue. (https://github.com/MichMich/MagicMirror/commit/e1fe8d1d89a96fd039ff5411d03aa3e5a3ac2452)

shbatm, 2019-01-03:
@MichMich Good to know. I was updating package.json just so I could do a clean `npm install`.

Electron 3.0.13 is working fine for me. The only thing was I had to update my Node.JS to v10.15.0 from v.8.15.0 -- I'm not sure if that's really required, but it looks like Electron 3.x is uses NODE_MODULE_VERSION 64 instead of 57, and it was causing a few of my modules to fail rebuilds (electron-rebuild was choking).