MagicMirrorOrg/MagicMirror #1502: 2 processes running? MagicMirror and MM?

anthony6608, 2019-01-03:
Hi guys!

I noticed on my Rpi PM2 shows two processes running MagicMirror and mm .. is this normal? I ask because I notice the MagicMirror has been locking up a lot recently and I am forced to reboot. I am using the latest software as this was installed earlier in the week.

Appreciate all your help and sorry if this question was asked already!

MichMich, 2019-01-03:
One is the actual MagicMirror process. The other is the start script used by PM2.

closed 2019-01-03 by MichMich

anthony6608, 2019-01-03:
Thanks Mich

I ended up removing it it... and to me the system seems to be running a lot smoother. I actually had two magic mirror browser  files running when I checked the GUI. Would it cause any issues leaving it like this? Just in case how do I add it back?

Thanks again

vincep5, 2019-01-03:
I thought one process was from the automatic setup script and the other you may have manually added from : https://github.com/MichMich/MagicMirror/wiki/Auto-Starting-MagicMirror

anthony6608, 2019-01-03:
I’m not sure but I know the two instances were magicmirror and mm. When I removed and stopped mm I noticed my pi wasn’t crashing anymore. I thought it was two separate instances because when I minimized magicmirror and went to the desktop I could see two separate magic mirror windows. Since I’ve deleted one my pi hasn’t crashed and my apis aren’t being blocked. For example the nest module kept getting blocked because of this I think. Now I’m no longer being blocked. Can someone help me figure out what’s going on? Thanks!

anthony6608, 2019-01-03:
Also when I restart my pi without mm running everything starts fine. I may have missed something somewhere but cant seem to figure out where and I want to make sure i have everything running fine. Currently this is what pm2 list outputs and my magicmirror is showing on the screen.

pi@raspberrypi:~/MagicMirror/config $ pm2 list
┌─────────────┬────┬─────────┬──────┬──────┬────────┬─────────┬────────┬─────┬──────────┬──────┬──────────┐
│ App name    │ id │ version │ mode │ pid  │ status │ restart │ uptime │ cpu │ mem      │ user │ watching │
├─────────────┼────┼─────────┼──────┼──────┼────────┼─────────┼────────┼─────┼──────────┼──────┼──────────┤
│ MagicMirror │ 0  │ 2.6.0   │ fork │ 1571 │ online │ 1       │ 13h    │ 0%  │ 1.5 MB   │ pi   │ enabled  │
└─────────────┴────┴─────────┴──────┴──────┴────────┴─────────┴────────┴─────┴──────────┴──────┴──────────┘
 Use `pm2 show <id|name>` to get more details about an app
pi@raspberrypi:~/MagicMirror/config $

anthony6608, 2019-01-03:
Just in case, heres the output from pm2 logs:

pi@raspberrypi:~/MagicMirror/config $ pm2 logs
[TAILING] Tailing last 15 lines for [all] processes (change the value with --lines option)
/home/pi/.pm2/pm2.log last 15 lines:
PM2        | 2019-01-02T22:31:34: PM2 log: pid=826 msg=failed to kill - retrying in 100ms
PM2        | 2019-01-02T22:31:34: PM2 log: pid=826 msg=failed to kill - retrying in 100ms
PM2        | 2019-01-02T22:31:34: PM2 log: pid=826 msg=failed to kill - retrying in 100ms
PM2        | 2019-01-02T22:31:34: PM2 log: pid=826 msg=failed to kill - retrying in 100ms
PM2        | 2019-01-02T22:31:34: PM2 log: pid=826 msg=failed to kill - retrying in 100ms
PM2        | 2019-01-02T22:31:34: PM2 log: pid=826 msg=failed to kill - retrying in 100ms
PM2        | 2019-01-02T22:31:34: PM2 log: pid=826 msg=failed to kill - retrying in 100ms
PM2        | 2019-01-02T22:31:34: PM2 log: pid=826 msg=failed to kill - retrying in 100ms
PM2        | 2019-01-02T22:31:34: PM2 log: pid=826 msg=failed to kill - retrying in 100ms
PM2        | 2019-01-02T22:31:34: PM2 log: pid=826 msg=failed to kill - retrying in 100ms
PM2        | 2019-01-02T22:31:35: PM2 log: Process with pid 826 still alive after 1600ms, sending it SIGKILL now...
PM2        | 2019-01-02T22:31:35: PM2 log: App [MagicMirror:0] exited with code [0] via signal [SIGKILL]
PM2        | 2019-01-02T22:31:35: PM2 log: pid=826 msg=process killed
PM2        | 2019-01-02T22:31:35: PM2 log: App [MagicMirror:0] starting in -fork mode-
PM2        | 2019-01-02T22:31:35: PM2 log: App [MagicMirror:0] online

/home/pi/.pm2/logs/MagicMirror-error.log last 15 lines:
0|MagicMir | npm ERR! 
0|MagicMir | npm ERR! Failed at the magicmirror@2.6.0 start script.
0|MagicMir | npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
0|MagicMir | 
0|MagicMir | npm ERR! A complete log of this run can be found in:
0|MagicMir | npm ERR!     /home/pi/.npm/_logs/2019-01-03T03_27_21_703Z-debug.log
0|MagicMir | libGL error: MESA-LOADER: failed to retrieve device information
0|MagicMir | MESA-LOADER: failed to retrieve device information
0|MagicMir | MESA-LOADER: failed to retrieve device information
0|MagicMir | libGL error: MESA-LOADER: failed to retrieve device information
0|MagicMir | MESA-LOADER: failed to retrieve device information
0|MagicMir | MESA-LOADER: failed to retrieve device information
0|MagicMir | libGL error: MESA-LOADER: failed to retrieve device information
0|MagicMir | MESA-LOADER: failed to retrieve device information
0|MagicMir | MESA-LOADER: failed to retrieve device information

/home/pi/.pm2/logs/MagicMirror-out.log last 15 lines:
0|MagicMir | Connecting socket for: MMM-GoogleMapsTraffic
0|MagicMir | Starting node_helper for module: MMM-GoogleMapsTraffic
0|MagicMir | Connecting socket for: MMM-MyCommute
0|MagicMir | ====================== Starting node_helper for module [MMM-MyCommute]
0|MagicMir | Connecting socket for: calendar
0|MagicMir | Starting node helper for: calendar
0|MagicMir | Connecting socket for: updatenotification
0|MagicMir | Connecting socket for: newsfeed
0|MagicMir | Starting module: newsfeed
0|MagicMir | Sockets connected & modules started ...
0|MagicMir | Launching application.
0|MagicMir | Create new calendar fetcher for url: http://www.calendarlabs.com/templates/ical/US-Holidays.ics - Interval: 300000
0|MagicMir | Working notification system. Notification: MMM-GOOGLE_MAPS_TRAFFIC-GET payload:  { style: 'black' }
0|MagicMir | Create new calendar fetcher for url: http://localhost:8080/modules/calendars/0243663de2ca1299df2e0f81170b06455bb3a8451cfffc8d9f046d311ff837cf.ics - Interval: 300000
0|MagicMir | Create new news fetcher for url: http://www.nytimes.com/services/xml/rss/nyt/HomePage.xml - Interval: 300000