MagicMirrorOrg/MagicMirror #1503: Change weatherforecast and currentweather unit display

fwitte, 2019-01-03:
Hi everybody,
I just started building/configuring my own MM and found a bug using "default" for the units. The degree symbol should not be displayed. It is degree Celsius and degree Fahrenheit, but not degree Kelvin.

**weatherforecast**
https://github.com/MichMich/MagicMirror/blob/de57daa3cd881ce1a14b88307bf61e8109879c81/modules/default/weatherforecast/weatherforecast.js#L145-L158

**currentweather**
https://github.com/MichMich/MagicMirror/blob/de57daa3cd881ce1a14b88307bf61e8109879c81/modules/default/currentweather/currentweather.js#L200-L213

https://github.com/MichMich/MagicMirror/blob/de57daa3cd881ce1a14b88307bf61e8109879c81/modules/default/currentweather/currentweather.js#L221

I suggest changing the display for both modules like this:

```
var degreeLabel = "";
if(this.config.scale) {
	switch(this.config.units) {
	case "metric":
		degreeLabel += "&deg; C";
		break;
	case "imperial":
		degreeLabel += "&deg; F";
		break;
	case "default":
		degreeLabel = " K";
		break;
	}
}
```

**currentweather**
```
temperature.innerHTML = " " + this.temperature.replace(".", this.config.decimalSymbol) + degreeLabel;
```

**weatherforecast**
```
maxTempCell.innerHTML = forecast.maxTemp.replace(".", this.config.decimalSymbol) + degreeLabel;
minTempCell.innerHTML = forecast.minTemp.replace(".", this.config.decimalSymbol) + degreeLabel;
```

If these changes are okay with you, I will open two PR for this issue.

fewieden, 2019-01-03:
I will also change it for the new weather module.

fewieden, 2019-01-03:
Just checked you can disable it in the new weather module by specifying config option `degreeLabel: false`

fewieden, 2019-01-05:
Handled in https://github.com/MichMich/MagicMirror/pull/1510 for the new weather module.