MagicMirrorOrg/MagicMirror #1508: major errors while trying to install MM 2.6.0...

justjim1220, 2019-01-04:
jiman@DESKTOP-LNUAHO0 MINGW64 ~/MagicMirror (master)                                                                      $ npm i                                                                                                                   npm WARN deprecated time-grunt@2.0.0: Deprecated because Grunt is practically un                                          maintained. Move on to something better. This package will continue to work with                                           Grunt v1, but it will not receive any updates.                                                                           npm WARN deprecated nomnom@1.8.1: Package no longer supported. Contact support@n                                          pmjs.com for more info.                                                                                                                                                                                                                             > electron-chromedriver@1.8.0 install C:\Users\jiman\MagicMirror\node_modules\el                                          ectron-chromedriver                                                                                                       > node ./download-chromedriver.js                                                                                                                                                                                                                   successfully dowloaded and extracted!                                                                                                                                                                                                               > clarinet@0.12.2 postinstall C:\Users\jiman\MagicMirror\node_modules\clarinet                                            > cd benchmark && npm i                                                                                                                                                                                                                             npm WARN deprecated jade@0.26.3: Jade has been renamed to pug, please install th                                          e latest version of pug instead of jade                                                                                                                                                                                                             > clarinet@0.12.2 postinstall C:\Users\jiman\MagicMirror\node_modules\clarinet\b                                          enchmark\node_modules\clarinet                                                                                            > cd benchmark && npm i                                                                                                                                                                                                                             npm WARN deprecated jade@0.26.3: Jade has been renamed to pug, please install th                                          e latest version of pug instead of jade                                                                                                                                                                                                             > clarinet@0.12.2 postinstall C:\Users\jiman\MagicMirror\node_modules\clarinet\b                                          enchmark\node_modules\clarinet\benchmark\node_modules\clarinet                                                            > cd benchmark && npm i                                                                                                                                                                                                                             npm WARN deprecated jade@0.26.3: Jade has been renamed to pug, please install th                                          e latest version of pug instead of jade                                                                                                                                                                                                             > clarinet@0.12.2 postinstall C:\Users\jiman\MagicMirror\node_modules\clarinet\b                                          enchmark\node_modules\clarinet\benchmark\node_modules\clarinet\benchmark\node_mo                                          dules\clarinet                                                                                                            > cd benchmark && npm i                                                                                                                                                                                                                             npm WARN deprecated jade@0.26.3: Jade has been renamed to pug, please install th                                          e latest version of pug instead of jade                                                                                                                                                                                                             > clarinet@0.12.2 postinstall C:\Users\jiman\MagicMirror\node_modules\clarinet\b                                          enchmark\node_modules\clarinet\benchmark\node_modules\clarinet\benchmark\node_mo                                          dules\clarinet\benchmark\node_modules\clarinet                                                                            > cd benchmark && npm i                                                                                                                                                                                                                             npm WARN deprecated jade@0.26.3: Jade has been renamed to pug, please install th                                          e latest version of pug instead of jade                                                                                                                                                                                                             > clarinet@0.12.2 postinstall C:\Users\jiman\MagicMirror\node_modules\clarinet\b                                          enchmark\node_modules\clarinet\benchmark\node_modules\clarinet\benchmark\node_mo                                          dules\clarinet\benchmark\node_modules\clarinet\benchmark\node_modules\clarinet                                            > cd benchmark && npm i                                                                                                                                                                                                                             npm WARN deprecated jade@0.26.3: Jade has been renamed to pug, please install th                                          e latest version of pug instead of jade                                                                                                                                                                                                             > clarinet@0.12.2 postinstall C:\Users\jiman\MagicMirror\node_modules\clarinet\b                                          enchmark\node_modules\clarinet\benchmark\node_modules\clarinet\benchmark\node_mo                                          dules\clarinet\benchmark\node_modules\clarinet\benchmark\node_modules\clarinet\b                                          enchmark\node_modules\clarinet                                                                                            > cd benchmark && npm i                                                                                                                                                                                                                             npm WARN deprecated jade@0.26.3: Jade has been renamed to pug, please install th                                          e latest version of pug instead of jade                                                                                                                                                                                                             > clarinet@0.12.2 postinstall C:\Users\jiman\MagicMirror\node_modules\clarinet\b                                          enchmark\node_modules\clarinet\benchmark\node_modules\clarinet\benchmark\node_mo                                          dules\clarinet\benchmark\node_modules\clarinet\benchmark\node_modules\clarinet\b   [... cut ...] output above.                                                                                                                                                                                                                                  npm ERR! A complete log of this run can be found in:                                                                      npm ERR!     C:\Users\jiman\AppData\Roaming\npm-cache\_logs\2019-01-04T22_38_35_                                          836Z-debug.log
--

khassel, 2019-01-05:
had similar problems yesterday but now its working (without any changes) ... may a temporary problem with the node.js stuff.

jackwilsdon, 2019-01-05:
Looks like this was an issue with the `clarinet` package (https://github.com/dscape/clarinet/issues/58) which should be fixed now (`0.12.3` has been published). Could you try this again @justjim1220?

swvalenti194, 2019-01-06:
Trying to update and this is what I get...any ideas?

pi@raspberrypi:~/MagicMirror $ git pull && npm install
Updating 6db61b4..de57daa
error: Your local changes to the following files would be overwritten by merge:
	package-lock.json
	vendor/package-lock.json
Please commit your changes or stash them before you merge.
Aborting

MichMich, 2019-01-08:
Please refer to http://forum.magicmirror.builders for support and troubleshooting.

I'm closing this issue for now. If you feel this issue should remain open, please let me know.

closed 2019-01-08 by MichMich

justjim1220, 2019-01-08:
It's all good now, It decided to work, and I didn't change anything. Not sure what was going on, but I kept getting the errors with both my Windows and my Ubuntu MM's. Both are working now!