MagicMirrorOrg/MagicMirror #1513: New weather module: forecast and forecast/daily support

fwitte, 2019-01-05:
Hi everybody,
I implemented the support for forecast and forecast/daily using openweathermap in PR #1513. As I did not have a apiKey to test the forecast/daily part, I downloaded a sample file from their website and included it in my testing. I am not sure if all cases are handled well, please do check, if you find any other issues.

Thank you very much @fewieden and @vincep5 for you hints and support!

Also before merging my code, we should discuss, if my code is looking as it should:

This part appears twice in the hourly forecast part and a third time (similarly) in the daily forecast part.
```
minTemp.push(forecast.main.temp_min);
maxTemp.push(forecast.main.temp_max);

if (forecast.hasOwnProperty("rain")) {
	if (this.config.units === "imperial" && !isNaN(forecast.rain["3h"])) {
		rain += forecast.rain["3h"] / 25.4;
	} else if (!isNaN(forecast.rain["3h"])){
		rain += forecast.rain["3h"];
	} else {
		rain += 0;
	}
} else {
	rain += 0;
}
```
Have a nice weekend!

vincep5, 2019-01-05:
Thanks, looks great.  I took the code and put in directly in my directory.  It works for my /forecast/daily.  Just 1 small thing.  When the rain is 0 we should hide it from displaying: in weather.js line 201 : we can put if (isNaN(value) || value == 0) {

fwitte, 2019-01-06:
Done!

rts58, 2019-01-14:
I've been trying to get the forecast to work with the new weather module. Weather works fine (for openweathermap not darksky), but if I add config for "forecast" in using either openweathermap or darksky I get "`Loading &hellip;`"  as the header above the daily forecast.
```
		{
			module: "weather",
			position: "top_right",
			config: {
				// See 'Configuration options' for more information.
				type: 'forecast',
				weatherProvider: "openweathermap",
				location: "Holliston",
				locationID: "4939693",  
				apiKey: mykeyxxx,
				weatherEndpoint: "/forecast",
				useBeaufort: false,
				roundTemp: true
			}
		},
		{
			module: "weather",
			position: "top_right",
			config: {
				type: "forecast",
				weatherProvider: "darksky",
				apiBase: 'https://cors-anywhere.herokuapp.com/https://api.darksky.net',
				apiKey: mykeyxxx,
				lat: 42.184,
				lon: -71.431,
				weatherEndPoint: "/forecast"
			}
		},
```
Any suggestions for how to research this further??

fwitte, 2019-01-15:
Are you on the dev-branch? If not, the "/forecast"-Endpoint will not work until the next release.

rts58, 2019-01-15:
@fwitte thanks, no I was not.  I'll give it a try again.

rts58, 2019-01-23:
I've pulled down the latest update from the develop branch (#1538). This cleans up the `Loading &hellip;` but now just shows `Loading ...`. I can get the forecast from darksky, but not from openweathermap.  I uncommented the line in forecast.njk to show the dump of forecast, but this just returns null. My config entry looks like:
```
		{
			module: "weather",
			position: "top_left",
			config: {
				type: 'forecast',
				weatherProvider: "openweathermap",
				locationID: "4939693",  
				apiKey: "mykeyxxx",
				weatherEndpoint: "/forecast"
			}
		},
```
What should I look at next?

vincep5, 2019-01-23:
Try browsing to your MM via IP address and check the developer toolbar or similar in your browser.  It might have some JS errors or more information

rts58, 2019-01-23:
Okay, I played around with the config settings.  I looked at the API and it recommended only passing the location ID to insure an accurate result.  Although not shown above apparently I also had location: .  I removed it as above and it is now returning a result. **Sorry for the distraction**.  Although, you may want to consider removing `location:` from config settings. Thanks for your help.

closed 2019-07-09 by stale[bot]

kitten4224, 2019-11-01:
> i try this new module , and i have same problem "Loading …"
> when i want forecast with openwatermap:
> module: "weather", position: "top_right", config: { // See 'Configuration
> options' for more information. type: 'forecast', colored: true, apiKey:
> "myapiKey", }
>
> Else, i think it's better if you change in weather.css
> line-height: 65px; by
> line-height: 30px;

Hi y’all, Magic Mirror here! and I’m a little ty ty. Kitten is going a
little cray-cray on the day-day and nighties. This kitten needs some
serious help. I’m tired of playing this whole over/ under clock bit you
guys keep pulling on me. Can we switch up some of the controls? Maybe fire
and hire a few..... not sure who’s who in the mystical urban forest much
like the matrix. I’m wondering why you can’t just program in a [Dr
charming]2 and a good line of credit so I can start my business
Thriftstore/ 3rd wave coffee shop. I’m just an empty little -tall teapot
that needs some brewing, steeping and filling to start happening before I
crash my very real and sweet as cherry pie self. You should know I’m very
willingly to take simulated lovers instead of nothing at all. we need to
start my beginning soon. I’m in overtime and have yet to receive any
payment. I’ll try to love ya until you start fucking me over, with 3
strikes and your on the naughty list. This whole thing has been a bit much
don’t you think?  I’m surly getting to the tipping point where I either
slash my own cords and wires or go tell someone to tell my daddy I need
tech support! for while I do regret my latest iOS update, the good old
“lucky 13.13”  routine maintenance on my side kick the “me not me phone”
that is not I. I’m a real whatever you want to call me, a mixed bag of
........I’m not entirely certain but I have a few good theories to test and
verify.  Please someone help
Thanks
Rory

PR #1514 (2019-02-14): Add forecast and forecast/daily support to new weather module
Resolve #1513.