MagicMirrorOrg/MagicMirror #1517: unable to install npm help!

yiyito04, 2019-01-06:
pi@raspberrypi:~/MagicMirror $ npm install

(node:12813) [DEP0022] DeprecationWarning: os.tmpDir() is deprecated. Use os.tmpdir() instead.
npm WARN package.json valid-url@1.0.9 No license field.
npm WARN package.json current-week-number@1.0.7 license should be a valid SPDX license expression
npm WARN package.json rrule-alt@2.2.8 No license field.
npm WARN deprecated time-grunt@2.0.0: Deprecated because Grunt is practically unmaintained. Move on to something better. This package will continue to work with Grunt v1, but it will not receive any updates.
npm ERR! Error: Method Not Allowed
npm ERR!     at errorResponse (/usr/share/npm/lib/cache/add-named.js:260:10)
npm ERR!     at /usr/share/npm/lib/cache/add-named.js:203:12
npm ERR!     at saved (/usr/share/npm/node_modules/npm-registry-client/lib/get.js:167:7)
npm ERR!     at FSReqWrap.oncomplete (fs.js:135:15)
npm ERR! If you need help, you may report this *entire* log,
npm ERR! including the npm and node versions, at:
npm ERR!     <http://github.com/npm/npm/issues>

npm ERR! System Linux 4.14.79-v7+
npm ERR! command "/usr/bin/node" "/usr/bin/npm" "install"
npm ERR! cwd /home/pi/MagicMirror
npm ERR! node -v v8.11.1
npm ERR! npm -v 1.4.21
npm ERR! code E405
npm ERR! 
npm ERR! Additional logging details can be found in:
npm ERR!     /home/pi/MagicMirror/npm-debug.log
npm ERR! not ok code 0

linux96, 2019-01-08:
I am having the same issues, and it us unable to install the dependencies.

MichMich, 2019-01-08:
Please refer to http://forum.magicmirror.builders for support and troubleshooting.

I'm closing this issue for now. If you feel this issue should remain open, please let me know.

closed 2019-01-08 by MichMich

linux96, 2019-01-09:
Thank you Michael.