MagicMirrorOrg/MagicMirror #1520: new weather module: feature suggestion -> forecast plot

fwitte, 2019-01-07:
Hi guys,
instead of the tabular forecast look I would like to have a plot of the forecast data. Is this something more users might like, thus could be implemented into the weather module?

This is, how my first try looks like (for the code see the features/forecast_plot branch in my repository):

[image]

As I am lacking experience in webdesign and .js, there are still some concerns about my code:

- The Plotly-library is not loaded from the node-modules (`"https://cdn.plot.ly/plotly-1.2.0.min.js"`).
- From what I understand, the Plotly object must be pushed into a div-container with a specific id. I abused my weather container `<div id="module_7_weather">` for this purpose, which is a terrible workaround. This is the reason, why the location information is below the plot and overlaps with it. I tried adding an additional container (e .g. `<div id="forecast_plot" ></div>`) within the forecast.njk-file, but on reload the console returns, that the container was not found. I assume, it is not created before some first run?
- Width, height and colors of the Plot are hard coded. For a good solution these should be referenced to the styles from the .css-files.
- Precipitation information is not implemented at the moment.
- Y-Axis title would be more beautiful if displayed like "Temperature in °C" or "Temperature in °F" etc.. This would double the amount of code used in the weather.js to determine the displays for units according to the user specifications. Maybe the units for the different systems could be hard coded in some place and all modules could reference that place.

If some people are interested into this feature, I am happy to work on this together with you!

fewieden, 2019-01-07:
The problem with those kind of libraries is when the module dom updates. For every rerender you need to reinstantiate the plot.

Also you're using a really few of the features of this library, why not using vanilla js with canvas like http://www.tutorialspark.com/html5/HTML5_Canvas_dataplot_Demo.php instead?

MichMich, 2019-01-08:
As much as I love the plot of the weather data, I think this is better suited to be available as a third party module. The current module is difficult enough to maintain as it is. Adding visuals like this would make it even more difficult. Feel free to develop and release your own module.

I'll close this issue for now, If you feel it should remain open, feel free to let me know.

closed 2019-01-08 by MichMich