MagicMirrorOrg/MagicMirror #1524: npm installation ERROR

lkthomas, 2019-01-09:
pi@raspberrypi:~ $ bash -c "$(curl -sL https://raw.githubusercontent.com/MichMich/MagicMirror/master/installers/raspberry.sh)"

$$\      $$\                     $$\           $$\      $$\ $$\                                          $$$$$$\
$$$\    $$$ |                    \__|          $$$\    $$$ |\__|                                        $$  __$$\
$$$$\  $$$$ | $$$$$$\   $$$$$$\  $$\  $$$$$$$\ $$$$\  $$$$ |$$\  $$$$$$\   $$$$$$\   $$$$$$\   $$$$$$\  \__/  $$ |
$$\$$\$$ $$ | \____$$\ $$  __$$\ $$ |$$  _____|$$\$$\$$ $$ |$$ |$$  __$$\ $$  __$$\ $$  __$$\ $$  __$$\  $$$$$$  |
$$ \$$$  $$ | $$$$$$$ |$$ /  $$ |$$ |$$ /      $$ \$$$  $$ |$$ |$$ |  \__|$$ |  \__|$$ /  $$ |$$ |  \__|$$  ____/
$$ |\$  /$$ |$$  __$$ |$$ |  $$ |$$ |$$ |      $$ |\$  /$$ |$$ |$$ |      $$ |      $$ |  $$ |$$ |      $$ |
$$ | \_/ $$ |\$$$$$$$ |\$$$$$$$ |$$ |\$$$$$$$\ $$ | \_/ $$ |$$ |$$ |      $$ |      \$$$$$$  |$$ |      $$$$$$$$\
\__|     \__| \_______| \____$$ |\__| \_______|\__|     \__|\__|\__|      \__|       \______/ \__|      \________|
                       $$\   $$ |
                       \$$$$$$  |
                        \______/

Updating packages ...
Hit:1 http://archive.raspberrypi.org/debian stretch InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Reading package lists... Done
Installing helper tools ...
Reading package lists... Done
Building dependency tree       
Reading state information... Done
build-essential is already the newest version (12.3).
curl is already the newest version (7.52.1-5+deb9u8).
git is already the newest version (1:2.11.0-3+deb9u4).
unzip is already the newest version (6.0-21).
wget is already the newest version (1.18-5+deb9u2).
0 upgraded, 0 newly installed, 0 to remove and 41 not upgraded.
Check current Node installation ...
Node currently installed. Checking version number.
Minimum Node version: v5.1.0
Installed Node version: v8.11.1
No Node.js upgrade necessary.
Cloning MagicMirror ...
Cloning into 'MagicMirror'...
remote: Enumerating objects: 302, done.
remote: Counting objects: 100% (302/302), done.
remote: Compressing objects: 100% (270/270), done.
remote: Total 302 (delta 37), reused 171 (delta 22), pack-reused 0
Receiving objects: 100% (302/302), 614.30 KiB | 0 bytes/s, done.
Resolving deltas: 100% (37/37), done.
Cloning MagicMirror Done!
Installing dependencies ...
(node:5846) [DEP0022] DeprecationWarning: os.tmpDir() is deprecated. Use os.tmpdir() instead.
npm WARN deprecated time-grunt@2.0.0: Deprecated because Grunt is practically unmaintained. Move on to something better. This package will continue to work with Grunt v1, but it will not receive any updates.
npm WARN deprecated express-ipfilter@0.3.1: This package is no longer being maintained and contains security vulnerabilities
npm ERR! registry error parsing json
npm ERR! SyntaxError: Unexpected token < in JSON at position 0
npm ERR! <!DOCTYPE html>
npm ERR! 
npm ERR! 
npm ERR! 
npm ERR!  <html class="no-js" lang="en-US"> 
npm ERR! <head>
npm ERR! <title>Could not find host | registry.npmjs.org | Cloudflare</title>
npm ERR! <meta charset="UTF-8" />
npm ERR! <meta http-equiv="Content-Type" content="text/html; charset=UTF-8" />
npm ERR! <meta http-equiv="X-UA-Compatible" content="IE=Edge,chrome=1" />
npm ERR! <meta name="robots" content="noindex, nofollow" />
npm ERR! <meta name="viewport" content="width=device-width,initial-scale=1,maximum-scale=1" />
npm ERR! <link rel="stylesheet" id="cf_styles-css" href="/cdn-cgi/styles/cf.errors.css" type="text/css" media="screen,projection" />
npm ERR! 
npm ERR! <style type="text/css">body{margin:0;padding:0}</style>
npm ERR! 
npm ERR! 
npm ERR! <script type="text/javascript" src="/cdn-cgi/scripts/zepto.min.js"></script>
npm ERR! <script type="text/javascript" src="/cdn-cgi/scripts/cf.common.js"></script>
npm ERR! 
npm ERR! 
npm ERR! 
npm ERR! </head>
npm ERR! <body>
npm ERR!   <div id="cf-wrapper">
npm ERR!     <div class="cf-alert cf-alert-error cf-cookie-error" id="cookie-alert" data-translate="enable_cookies">Please enable cookies.</div>
npm ERR!     <div id="cf-error-details" class="cf-error-details-wrapper">
npm ERR!       <div class="cf-wrapper cf-header cf-error-overview">
npm ERR!         <h1>
npm ERR!           <span class="cf-error-type" data-translate="error">Error</span>
npm ERR!           <span class="cf-error-code">1018</span>
npm ERR!           <small class="heading-ray-id">Ray ID: 49667c04fedb32c5 &bull; 2019-01-09 11:15:15 UTC</small>
npm ERR!         </h1>
npm ERR!         <h2 class="cf-subheadline">Could not find host</h2>
npm ERR!       </div>
npm ERR! 
npm ERR!       <section></section>
npm ERR! 
npm ERR!       <div class="cf-section cf-wrapper">
npm ERR!         <div class="cf-columns two">
npm ERR!           <div class="cf-column">
npm ERR!             <h2 data-translate="what_happened">What happened?</h2>
npm ERR!             <p>You've requested a page on a website (registry.npmjs.org) that is on the <a data-orig-proto="https" data-orig-ref="www.cloudflare.com/5xx-error-landing?utm_source=error_100x" target="_blank">Cloudflare</a> network. Cloudflare is currently unable to resolve your requested domain (registry.npmjs.org). There are two potential causes of this:</p>
npm ERR!             <ul>
npm ERR!                <li><strong>Most likely:</strong> if the owner just signed up for Cloudflare it can take a few minutes for the website's information to be distributed to our global network.</li>
npm ERR!                <li><strong>Less likely:</strong> something is wrong with this site's configuration. Usually this happens when accounts have been signed up with a partner organization (e.g., a hosting provider) and the provider's DNS fails.</li>
npm ERR!             </ul>
npm ERR!           </div>
npm ERR! 
npm ERR!           
npm ERR!         </div>
npm ERR!       </div>
npm ERR! 
npm ERR!       <div class="cf-error-footer cf-wrapper">
npm ERR!   <p>
npm ERR!     <span class="cf-footer-item">Cloudflare Ray ID: <strong>49667c04fedb32c5</strong></span>
npm ERR!     <span class="cf-footer-separator">&bull;</span>
npm ERR!     <span class="cf-footer-item"><span>Your IP</span>: 119.246.2.238</span>
npm ERR!     <span class="cf-footer-separator">&bull;</span>
npm ERR!     <span class="cf-footer-item"><span>Performance &amp; security by</span> <a href="https://www.cloudflare.com/5xx-error-landing?utm_source=error_footer" id="brand_link" target="_blank">Cloudflare</a></span>
npm ERR!     
npm ERR!   </p>
npm ERR! </div>
npm ERR! 
npm ERR! 
npm ERR!     </div>
npm ERR!   </div>
npm ERR! 
npm ERR!   <script type="text/javascript">
npm ERR!   window._cf_translation = {};
npm ERR!   
npm ERR!   
npm ERR! </script>
npm ERR! 
npm ERR! </body>
npm ERR! </html>
npm ERR! 
npm ERR!     at JSON.parse (<anonymous>)
npm ERR!     at RegClient.<anonymous> (/usr/share/npm/node_modules/npm-registry-client/lib/request.js:266:23)
npm ERR!     at Request._callback (/usr/share/npm/node_modules/npm-registry-client/lib/request.js:238:65)
npm ERR!     at Request.self.callback (/usr/lib/nodejs/request/index.js:148:22)
npm ERR!     at emitTwo (events.js:126:13)
npm ERR!     at Request.emit (events.js:214:7)
npm ERR!     at Request.<anonymous> (/usr/lib/nodejs/request/index.js:896:14)
npm ERR!     at emitOne (events.js:121:20)
npm ERR!     at Request.emit (events.js:211:7)
npm ERR!     at IncomingMessage.<anonymous> (/usr/lib/nodejs/request/index.js:847:12)
npm ERR! If you need help, you may report this *entire* log,
npm ERR! including the npm and node versions, at:
npm ERR!     <http://github.com/npm/npm/issues>

npm ERR! System Linux 4.14.79-v7+
npm ERR! command "/usr/bin/node" "/usr/bin/npm" "install"
npm ERR! cwd /home/pi/MagicMirror
npm ERR! node -v v8.11.1
npm ERR! npm -v 1.4.21
npm ERR! registry error parsing json
npm ERR! registry error parsing json
npm ERR! fetch failed https://registry.npmjs.org/finalhandler/-/finalhandler-1.1.1.tgz
npm WARN deprecated nomnom@1.8.1: Package no longer supported. Contact support@npmjs.com for more info.
 
> electron-chromedriver@1.8.0 install /home/pi/MagicMirror/node_modules/spectron/node_modules/electron-chromedriver
> node ./download-chromedriver.js

successfully dowloaded and extracted!
npm WARN deprecated circular-json@0.3.3: CircularJSON is in maintenance only, flatted is its successor.
npm WARN deprecated browserslist@2.11.3: Browserslist 2 could fail on reading Browserslist >3.0 config used in other tools.
npm ERR! 
npm ERR! Additional logging details can be found in:
npm ERR!     /home/pi/MagicMirror/npm-debug.log
npm ERR! not ok code 0
Unable to install dependencies!
pi@raspberrypi:~ $

lkthomas, 2019-01-09:
anyone have an idea why? This is brand new image of RPi 3

MichMich, 2019-01-09:
It seems your PI could not reach registry.npmjs.org. It might be because it has connectivity issues or because NPM was down (in that case try again later). 

Please note that the issues here on GitHub are intended to report replicatable issues in the MagicMirror² core. Please refer to http://forum.magicmirror.builders for support and troubleshooting.

I'm closing this issue for now. If you feel this issue should remain open, please let me know.

closed 2019-01-09 by MichMich

lkthomas, 2019-01-09:
original error went away, another error happens,

pi@raspberrypi:~/MagicMirror $ npm install
(node:6315) [DEP0022] DeprecationWarning: os.tmpDir() is deprecated. Use os.tmpdir() instead.
npm WARN package.json current-week-number@1.0.7 license should be a valid SPDX license expression
npm WARN package.json rrule-alt@2.2.8 No license field.
npm WARN package.json valid-url@1.0.9 No license field.
npm WARN deprecated time-grunt@2.0.0: Deprecated because Grunt is practically unmaintained. Move on to something better. This package will continue to work with Grunt v1, but it will not receive any updates.
npm ERR! Error: Method Not Allowed
npm ERR!     at errorResponse (/usr/share/npm/lib/cache/add-named.js:260:10)
npm ERR!     at /usr/share/npm/lib/cache/add-named.js:203:12
npm ERR!     at saved (/usr/share/npm/node_modules/npm-registry-client/lib/get.js:167:7)
npm ERR!     at FSReqWrap.oncomplete (fs.js:135:15)
npm ERR! If you need help, you may report this *entire* log,
npm ERR! including the npm and node versions, at:
npm ERR!     <http://github.com/npm/npm/issues>

npm ERR! System Linux 4.14.79-v7+
npm ERR! command "/usr/bin/node" "/usr/bin/npm" "install"
npm ERR! cwd /home/pi/MagicMirror
npm ERR! node -v v8.11.1
npm ERR! npm -v 1.4.21
npm ERR! code E405
npm ERR! 
npm ERR! Additional logging details can be found in:
npm ERR!     /home/pi/MagicMirror/npm-debug.log
npm ERR! not ok code 0

does it mean I have to try again later? Sorry, I am new to MagicMirror project, and this is my first deployment.

MichMich, 2019-01-09:
Not familiar with this error, please open a support topic on http://forum.magicmirror.builders. Thanks!

mihanich, 2019-01-31:
> original error went away, another error happens,
>
> pi@raspberrypi:~/MagicMirror $ npm install
> (node:6315) [DEP0022] DeprecationWarning: os.tmpDir() is deprecated. Use os.tmpdir() instead.
> npm WARN [...]

try:
`sudo npm i -g npm`
This fix error for me.

Vaishnavikumara, 2019-07-10:
Hi @mihanich , I tried what you said but this is what I got 
npm WARN deprecated time-grunt@2.0.0: Deprecated because Grunt is practically un  maintained. Move on to something better. This package will continue to work with   Grunt v1, but it will not receive any updates.
npm WARN registry Unexpected warning for https://registry.npmjs.org/: Miscellane  ous Warning ENODATA: Cache input stream was empty
npm WARN registry Using stale data from https://registry.npmjs.org/ due to a req  uest error during revalidation.
npm ERR! code EINTEGRITY
npm ERR! errno EINTEGRITY
npm ERR! Invalid response body while trying to fetch https://registry.npmjs.org/  @babel%2fcore: Integrity verification failed for sha512-awUDtusHAsJYo9NCrF9BoJ5v  MY6BcdWtvDMmdMfCQ+zkouN8yv1huE8kYJ6be892yav929G0HVhLWPyU+G+Xcw== (/home/pi/.npm/  _cacache/content-v2/sha512/6b/05/03b6eb0702c258a3d342ac5f41a09e6f318e8171d5adbc3  32674c7c243ece4a2e37ccafd61b84f24609e9b7bcf76c9abfddbd1b41d584b58fc94f86f9773)

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2019-07-10T04_13_51_651Z-debug.log
Unable to install dependencies!

please help

Ajimsutar, 2020-12-25:
> Hi @mihanich , I tried what you said but this is what I got
> npm WARN deprecated time-grunt@2.0.0: Deprecated because Grunt is practically un maintained. Move on to something better. This package [...]

I am having same problem if u found solution let me know

rejas, 2020-12-25:
What version are you using? Looks like yours is outdated....

Ajimsutar, 2020-12-25:
I am using Raspberry Pi 3b.can you provide me newer version of magicmirror

rejas, 2020-12-25:
You didnt answer my question: Which version of MagicMirror are you using?

Ajimsutar, 2020-12-25:
> You didnt answer my question: Which version of MagicMirror are you using?

I am using this command bash -c "$(curl -sL https://raw.githubusercontent.com/MichMich/MagicMirror/master/installers/raspberry.sh)"

rejas, 2020-12-25:
Please take a look at the updated installation isntruction: https://docs.magicmirror.builders/getting-started/installation.html

sdetweil, 2020-12-25:
@Ajimsutar the install script was removed a couple releases ago. I maintain it separately now

see https://github.com/sdetweil/MagicMirror_scripts

Ajimsutar, 2020-12-25:
While installing npm it gives error in cromedriver

sdetweil, 2020-12-25:
@Ajimsutar are u installing on pi0? electron doesn't exist for pi0 , that is just a warning, not a failure

Ajimsutar, 2020-12-25:
I am installing on pi3

sdetweil, 2020-12-25:
if u used my script, email me the ~/install.log
same userid on gmail

Ajimsutar, 2020-12-26:
now its giving error of cromedriver

sdetweil, 2020-12-26:
send me the log file

Ajimsutar, 2020-12-27:
This is log file

sdetweil, 2020-12-27:
sorry, I don't see a log file

Ajimsutar, 2020-12-27:
Can you send me email id

sdetweil, 2020-12-27:
its the same id  as I use here at gmail.

Ajimsutar, 2020-12-27:
Actually I couldn't find your email address can you send me please

sdetweil, 2020-12-27:
come on sdetweil at gmail   (you have to finish it).. this way the spam bots don't...

Ajimsutar, 2020-12-27:
Yes I have sent you log file

sdetweil, 2020-12-27:
thanks.. but that wasn't the log I was looking for.. 

i was looking for the log from my auto install script, which should be located at ~/install.log  
the script makes lot of decisions and does updates based on that..  

i see the failure you reported, but don't see the rest of the info I was looking for 

remove the MagicMirror folder and run the install script from here, one line cut and paste
https://github.com/sdetweil/MagicMirror_scripts

Ajimsutar, 2020-12-29:
i have send you new log file

sdetweil, 2020-12-30:
status. he had some problem with npm reinstalling npm resolved