MagicMirrorOrg/MagicMirror #1529: can only manully open mirror. I tried the mm thing in the pm2 thing and it didnt work. Also tried

Brazyb, 2019-01-12:
I tried the mm thing in the pm2 thing and it didn't work. Also,  I tried
sudo pm2 startup systemd -u pi --hp /home/pi
npm rebuild

**Platform**:raberry pi 3 b

**Node Version**: idk

**MagicMirror Version**:v2.

**Expected Results**: I want to type in pm2 start MagicMirror and have the program pop up 

**Actual Results**: if you can't see the picture it shows magic mirror online and mm stopped
if i go into the 
pm2 logs MagicMirror [--lines 1000]
it shows up as 

pi@raspberrypi:~ $ pm2 logs MagicMirror [--lines 1000]
[TAILING] Tailing last 15 lines for [MagicMirror] process (change the value with --lines option)
/home/pi/.pm2/logs/MagicMirror-out.log last 15 lines:
0|MagicMir | Whoops! There was an uncaught exception...
0|MagicMir | MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
0|MagicMir | If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues
0|MagicMir | Starting MagicMirror: v2.6.0
0|MagicMir | Whoops! There was an uncaught exception...
0|MagicMir | MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
0|MagicMir | If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues
0|MagicMir | Starting MagicMirror: v2.6.0
0|MagicMir | Whoops! There was an uncaught exception...
0|MagicMir | MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
0|MagicMir | If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues
0|MagicMir | Starting MagicMirror: v2.6.0
0|MagicMir | Whoops! There was an uncaught exception...
0|MagicMir | MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
0|MagicMir | If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues

[image]

[image]

fdahms, 2019-01-12:
try:
    sudo npm install -g pm2
    sudo pm2 startup systemd -u pi --hp /home/pi
    pm2 start ~/MagicMirror/installers/pm2_MagicMirror.json
    pm2 save

Brazyb, 2019-01-13:
the third line of code seems to not work. 
[image]

closed 2019-01-16 by Brazyb