MagicMirrorOrg/MagicMirror #1539: how do i stop mm to get my pi screen to work with it

lee-lvd, 2019-01-16:
Please only submit reproducible issues. 

If you're not sure if it's a real bug or if it's just you, please open a topic on the forum: [https://forum.magicmirror.builders/category/15/bug-hunt](https://forum.magicmirror.builders/category/15/bug-hunt)
Problems installing or configuring your MagicMirror? Check out: [https://forum.magicmirror.builders/category/10/troubleshooting](https://forum.magicmirror.builders/category/10/troubleshooting)

When submitting a new issue, please supply the following information:

**Platform**: Place your platform here... give us your web browser/Electron version *and* your hardware (Raspberry Pi 2/3, Windows, Mac, Linux, System V UNIX).

**Node Version**: Make sure it's version 0.12.13 or later.

**MagicMirror Version**: Now that the versions have split, tell us if you are using the PHP version (v1) or the newer JavaScript version (v2).

**Description**: Provide a detailed description about the issue and include specific details to help us understand the problem. Adding screenshots will help describing the problem.

**Steps to Reproduce**: List the step by step process to reproduce the issue.

**Expected Results**: Describe what you expected to see.

**Actual Results**: Describe what you actually saw.

**Configuration**: What does the used config.js file look like? Don't forget to remove any sensitive information!

**Additional Notes**: Provide any other relevant notes not previously mentioned. This is optional.

rts58, 2019-01-16:
I guess it depends upon how you started it.  If you have a keyboard connected you can alt-tab to a console window, if there is no prompt in the window use ctrl-c.  If there is you should be able to run `pm2 stop mm`.  You should be able to run this also connected via SSH. I found this [thread ](https://forum.magicmirror.builders/topic/4531/how-to-stop-magic-mirror-noob?page=1) in the forum which may help more.

ps I've only messed with this for a couple weeks, so not even a novice yet

ToddKlindt, 2019-01-16:
If you're on the box itself you can type Ctrl Q to close the browser and get back to the desktop.

closed 2019-01-17 by MichMich