MagicMirrorOrg/MagicMirror #1545: Show Snow amounts in new weather module

vincep5, 2019-01-18:
@fewieden @fwitte
We should implement showing Snow in the forecast for the new weather module.  I can work on this after the pending new weather PRs are merged.  

We should do one of these
1 - have a new config entry for showSnowAmount and show it separate from rain
2 - add both together as one precipitation amount

It looks like the darksky.js provider already takes precipAccumulation and puts the amount in the rain variable.  So rain and snow is already considered the same.
Dark Sky API :
`             "precipAccumulation": 7.337,
              "precipType": "snow",
`
darksky.js :
`weather.rain = forecast.precipAccumulation;`

The OpenWeatherMap API for forecast/daily does have "snow" which is a separate entry than "rain" 
We dont do anything with snow today :
`    "list": [
        {
            "dt": 1547834400,
            "temp": {
                "day": 31.05,
                "min": 25.83,
                "max": 31.05,
                "night": 28.62,
                "eve": 29.21,
                "morn": 29.19
            },
            "pressure": 1006.37,
            "humidity": 86,
            "weather": [
                {
                    "id": 601,
                    "main": "Snow",
                    "description": "snow",
                    "icon": "13d"
                }
            ],
            "speed": 4.61,
            "deg": 19,
            "clouds": 44,
            "snow": 7.79
        },
`

fwitte, 2019-01-23:
I remember, that the openweathermap-API (forecast, I did not check it with forecast/daily) provides both, rain and snow, simultaneously, if for instance it snows in the morning and rains in the evening. That would require an additional collumn in the table display. I am not 100 % sure about this, but maybe you should have a look into that! Have a nice day!

vincep5, 2019-01-23:
Thanks for the feedback.  Yes, correct, it can have both if the weather is like today for me.  Rain and snow mix !  forecast and forecast/daily both have rain and/or snow.  I have double checked that.  I guess what I could also implement, is if the icon is "rain" we show the rain amount and if the icon is "snow" we show the snow amount.  I'd have to look at all the different icon names to be sure.
I'm not sure if others would be in favor of adding another column to the display.   I would attempt to implement this in the same display column.

CriticalPoint, 2019-01-23:
> I would attempt to implement this in the same display column

As you said, it would be much better to flip the icons (weather dependent) as otherwise this may have undesired implications for anyone using custom CSS

CriticalPoint, 2019-01-24:
> Nowhere in their API do they have a rainfall amount.
> 

It looks like precipitation type is specified with `"precipType": "snow"`, or `"precipType": "rain"`, so just adding this to the route should yield the correct result.

vincep5, 2019-01-24:
Correct, but there isn't any other field having the rain amount in the Dark Sky API.  Dark Sky does more with rain probability % instead of amounts. I'll get a snippet of the JSON later

vincep5, 2019-01-24:
Here's a section from the Dark Sky JSON..
               `"time": 1548309600,
                "summary": "Mostly cloudy in the morning.",
                "icon": "partly-cloudy-day",
                "sunriseTime": 1548334163,
                "sunsetTime": 1548371887,
                "moonPhase": 0.63,
                "precipIntensity": 0.0005,
                "precipIntensityMax": 0.0043,
                "precipIntensityMaxTime": 1548316800,
                "precipProbability": 0.12,
                "precipType": "rain",
                "temperatureHigh": 46.95,
                "temperatureHighTime": 1548363600,
                "temperatureLow": 32.18,
                "temperatureLowTime": 1548414000,
                "apparentTemperatureHigh": 43.14,
                "apparentTemperatureHighTime": 1548363600,
                "apparentTemperatureLow": 27.93,
                "apparentTemperatureLowTime": 1548417600,
                "dewPoint": 31.91,
                "humidity": 0.8,
                "pressure": 1018.58,
                "windSpeed": 6.45,
                "windGust": 18.44,
                "windGustTime": 1548309600,
                "windBearing": 320,
                "cloudCover": 0.44,
                "uvIndex": 4,
                "uvIndexTime": 1548352800,
                "visibility": 10,
                "ozone": 297.25,
                "temperatureMin": 32.21,
                "temperatureMinTime": 1548338400,
                "temperatureMax": 46.95,
                "temperatureMaxTime": 1548363600,
                "apparentTemperatureMin": 26.01,
                "apparentTemperatureMinTime": 1548342000,
                "apparentTemperatureMax": 43.14,
                "apparentTemperatureMaxTime": 1548363600`

vincep5, 2019-02-14:
Pending PR https://github.com/MichMich/MagicMirror/pull/1573

closed 2019-02-17 by vincep5