MagicMirrorOrg/MagicMirror #1546: Location name is not returned in header

Mysh3ll, 2019-01-19:
Hi,

**fetchedLocatioName** is nerver assigned to the response data, so it will never append the header with the name of location.

https://github.com/MichMich/MagicMirror/blob/de57daa3cd881ce1a14b88307bf61e8109879c81/modules/default/currentweather/currentweather.js#L73

fewieden, 2019-01-20:
If you want that feature, you can tr the new default weather module, however it works only for the provider openweathermap.

PREngineer, 2019-01-28:
I cloned the repository recently and it doesn't append the location for me either on the default Current Weather module.

fewieden, 2019-02-27:
@PREngineer which provider do you use? It's not possible for darksky as the data doesn't contain the location.

If you use openweathermap, do you have a new or an old api key?

PREngineer, 2019-02-27:
I got my API Key last month from openweathermaps.

DazTypeR, 2019-03-17:
I too have this issue, installed latest version of Raspbian yesterday (Saturday 16th March 2019) on a brand new Pi 3B+, cloned the repository and created my API keys on the same date as above and the weather works but the location name is not displayed in the header (there is no header).

All modules are commented out apart from time / date, current weather and Google Photo's module.

API key was obtained from Openweathermaps as described in the install guide and the location ID was obtained from the Json file provided by Openweathermaps.

closed 2019-07-09 by stale[bot]