MagicMirrorOrg/MagicMirror #1548: showWindDirection: "false" doesn't work

nickbeaulieu98, 2019-01-21:
I don't want the wind direction to show in my weather module, so I've set showWindDirection: "false" but that doesn't do anything.

mattgahs, 2019-01-21:
Try `showWindDirection: false,` without the quotes around false.

nickbeaulieu98, 2019-01-21:
that did it. Thanks. I wouldn't have expected that, since all the other statements use quotes

closed 2019-01-21 by nickbeaulieu98

ToddKlindt, 2019-01-21:
> that did it. Thanks. I wouldn't have expected that, since all the other statements use quotes

I always struggle with this too. I've been using MM for a couple of years and I made this mistake just this last week. The trick I use is to look at the examples. The trues and falses are never in quotes, where the other config options are.

mattgahs, 2019-01-21:
I assume it's because true/false is a binary state, whereas text/numbers are strings that can have an infinite number of options that have to be defined by quotes.

fewieden, 2019-01-23:
@mattgahs numbers are normally also not in quotes, if there is a number in quotes it gets interpreted a text/string