MagicMirrorOrg/MagicMirror #1555: UpdateNotification fetch errors if github is unavailable

erdoukki, 2019-01-28:
UpdateNotification module made a lot of fetch error (DNS problems may locally be in trouble also)

Disabling UpdateNotiifcations may fix the issue.

---
fatal: unable to access 'https://github.com/MichMich/MagicMirror/': Could not resolve host: github.com

fatal: unable to access 'https://github.com/spectroman/mmm-moon-phases.git/': Could not resolve host: github.com

fatal: unable to access 'https://github.com/dmcinnes/MMM-forecast-io.git/': Could not resolve host: github.com

fatal: unable to access 'https://github.com/dmcinnes/MMM-forecast-io.git/': Could not resolve host: github.com

fatal: unable to access 'https://github.com/MichMich/MagicMirror/': Could not resolve host: github.com

fatal: unable to access 'https://github.com/spectroman/mmm-moon-phases.git/': Could not resolve host: github.com

fatal: unable to access 'https://github.com/MichMich/MagicMirror/': Could not resolve host: github.com

JohnB7777, 2019-01-29:
Hello,

I have the same error !
At the beginning, I thought it was because one of my module wasn't update but no.
And I have already my internet connection.
So... I don't understand.

MichMich, 2019-05-21:
Well ... yes. If there is o connection, there is an error. So it is reported. The number of errors could be decreased but other than that, it's expected behaviour. Feel free to send a PR to solve this.

Closing this issue for now.

closed 2019-05-21 by MichMich

theforceofhabit, 2021-06-04:
anyone found a solution yet?