MagicMirrorOrg/MagicMirror #1557: [Improvement] - Allow configuration of request path

kquinsland, 2019-01-30:
**Platform**:  

Firefox, OSX 10.14.4 & Android 9 & Windows 10. rPi; debian / docker

**Node Version**: 

```
# node --version
v10.5.0
```

**MagicMirror Version**: 

JS version.

**Description**: 
There is no apparent way to change the relative path that the magic-mirror server loads resources from. this makes it difficult to use magic mirror behind a multi-tenant proxy and a single `Host: ` header.

**Steps to Reproduce**: 

0. Install an instance of Magic Mirror on a container behind a proxy (like [traefik](https://traefik.io/))
1. Add other containers to the same host, also behind traefik
2. configure proxy to route traffic to backend containers based on the requested resource.
3. observe that magic mirror can not be configured to request resources from any base location other than `/`

**Expected Results**: 

I would expect a 
```
config = {
    base_path: "/mm/" # Defaults to `/` if not set
```
 or similar config directive which would configure magic mirror to load all resources from a common base path that is not `/`

**Additional Notes**:

The reason i'm asking for this feature is because it makes it very difficult to host MagicMirror behind a proxy.  E.G. I have several services (in containers...) running behind a proxy and i use the requested path to route requests to the proper container.  E.G.:  $service1 and mm are behind a proxy on $host1.  I would like to make GET requests to $host1/$service1 and $host1/mm and have the proxy forward the request to $service1 or $service2 depending on the path.

Currently, it looks like MM requests resources from several paths, all of which are in the ROOT of the server

- `/config/`
- `/vendor/`
- `/modules/`
- `/js/`
- `/translations/`
- `/fonts/`
- `/socket.io/`

and i don't see an easy way to configure or change this.

I would like a config directive that allows me to introduce a base-path that all needed resources will be relative to.  E.G.:

- `/$prefix/socket.io/`
- `/$prefix/fonts/`

... etc where `$prefix` is something that i can control so i can then configure my proxy to forward all requests to `/$prefix/*` to the magic mirror container.

Currently, i an working around this by creating two distinct hostnames that point to the same host and am using the `Host:` header to determine which container should receive the traffic.  I would like to use a single hostname / DNS entry, but can not due to Magic Mirror.

MichMich, 2019-01-30:
As long as it’s not a breaking change: feel free to send a PR. :)

kquinsland, 2019-01-30:
@MichMich I'd do just that... but i have no real idea where w/i  a node/express app one would go about doing so.  I haven't done a whole lot of JS development work since jQuery was hot.

Quick bit of google does not really show an easy way to do this.  I think? the change would need to go here: 
https://github.com/MichMich/MagicMirror/blob/master/js/server.js#L27

and all the setup here:
https://github.com/MichMich/MagicMirror/blob/master/js/server.js#L49

would also need some sort of "has the user set a value that's not default? prepend that..." logic

and i think? the similar `app.get()` calls (line 52, 56, 60) would need similar logic?

sdetweil, 2019-08-22:
looks like you could fix the global.root.path in js/app.js to be before MagicMirror
and then in js/server.js  use your prefix on the express paths (logical and physical)

closed 2019-10-28 by stale[bot]