MagicMirrorOrg/MagicMirror #1559: Improvement? - how to format "compliments" for quotes

feedbagg, 2019-02-02:
Hi!
I want to add some quotes im my compliments, therefore i'll add the name of the quoted person like this:

"Egal wie schwierig das Leben geworden ist. Es gibt immer etwas, worin du erfolgreich sein kannst.\NAMEStephen Hawking",

should look like:

[image]

may this be realizable?

Thank You in advance!
FeedBagg

MichMich, 2019-02-07:
Not in the current form, but feel free to send a PR. I'm not sure I like the /NAME solution. I think a better idea would be using a (json) object.

MichMich, 2019-02-14:
I think you are better of using one of the quotes modules. Please check: https://github.com/MichMich/MagicMirror/wiki/3rd-party-modules

closed 2019-02-14 by MichMich