MagicMirrorOrg/MagicMirror #1568: Calendar doesn't translate events to local time

rossburton, 2019-02-11:
My calendar has an event on it which is from 15:30 to 16:00 every week.  However it appears on my MM calendar as 07:30 to 08:00 because the event was created by a colleague in a different timezone: he's in PST (-0800) and I'm in GMT (+0000).

MichMich, 2019-02-14:
Can you try the dev branche? Might be fixed already.

narwhalsnarwhals, 2019-06-13:
it still isnt

mlcampbe, 2019-07-17:
I am having this problem too on the latest 2.8.0 release.

I am in CT and have colleagues in ET and PT. When an east coast person creates an event then it is off by 1 hours. If a pacific coast person creates an event it is off by 2 hours (the other direction). It is maddening and as a result I can't rely on the times.

Any ideas? Do you want a sample ical file to help debug it?

mlcampbe, 2019-07-17:
I may have found a quick fix for the problem. I did some research and found that the default calendar module uses the node-ical modules to parse calendar data. I notice that it is based off of https://github.com/peterbraden/ical.js. I also found https://github.com/jens-maus/node-ical that is based off of the original peterbraden/ical.js. but seems to have been updated to use the npm package moment-timezone.

I did a drop in replacement and it seems to work:

cd modules/default/calendar/vendor
mv ical.js ical.js.orig
git clone https://github.com/jens-maus/node-ical
mv node-ical/ ical.js
cd ical.js
npm install

Restarted MM and now my timezones are showing up properly. So I made my own modules out of this so it won't get overwritten by updates.

So it seems that the correct fix would be for MM to update the ical.js that it uses.

rossburton, 2019-07-17:
🏅 for @mlcampbe!

matthiashh, 2019-09-08:
I am having the same problem being on a31546b1ff8c7c10e12d7acbab063297d59c6b93.

Would there be an issue using the apparently updated version of the ical file?
i can create a PR if the proposed solution of @mlcampbe is in line with the projects policies.

DennisFaucher, 2020-02-06:
@mlcampbe Thank you so much. My Exchange calendar ICS was acting extremely funny. I think this may have fixed the issues.

MichMich, 2020-07-04:
Duplicate https://github.com/MichMich/MagicMirror/issues/1798, closing.

closed 2020-07-04 by MichMich