MagicMirrorOrg/MagicMirror #1579: Please help errno: 'EADDRINUSE',

anthony6608, 2019-02-18:
I know this has been discussed before but I can't seem to find a solution. 

About a week ago I performed a fresh install of MagicMirror on Raspberry pi running raspbian stretch. Immediately I noticed the default weather and calendar module wasn't displaying anything. Weather was enabled but wasn't showing up on the interface and no matter what I do calendar just shows "No upcoming events"...

 @sdetweil tried helping me in pm and suggested I run npm start dev and we noticed EADDRINUSE although the ip/port is not actually in use. Initially @sdetweil stated I had another process running but I can't seem to find it. I did change the default port from 8080 to 8081 and then reverted it back to default afterwards. Ive tried rebooting several times, killed electron processes and still nothing. I can't get rid of this error. Please help!

Connecting socket for: MMM-MTA
Connecting socket for: MMM-uber
Starting node helper for: MMM-uber
Connecting socket for: MMM-lyft
Starting node helper for: MMM-lyft
Connecting socket for: MMM-SystemStats
Connecting socket for: MMM-RTSPStream
Connecting socket for: newsfeed
Starting module: newsfeed
Sockets connected & modules started ...
Whoops! There was an uncaught exception...
{ Error: listen EADDRINUSE 127.0.0.1:8080
    at Object._errnoException (util.js:1024:11)
    at _exceptionWithHostPort (util.js:1046:20)
    at Server.setupListenHandle [as _listen2] (net.js:1351:14)
    at listenInCluster (net.js:1392:12)
    at GetAddrInfoReqWrap.doListen [as callback] (net.js:1501:7)
    at GetAddrInfoReqWrap.onlookup [as oncomplete] (dns.js:97:10)
  code: 'EADDRINUSE',
  errno: 'EADDRINUSE',
  syscall: 'listen',
  address: '127.0.0.1',
  port: 8080 }
MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues
Launching application.

NETSTAT OUTPUT: 

 netstat -lptn
(Not all processes could be identified, non-owned process info
 will not be shown, you would have to be root to see it all.)
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name    
tcp        0      0 127.0.0.1:8080          0.0.0.0:*               LISTEN      851/electron.js     
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN      -                   
tcp        0      0 0.0.0.0:5055            0.0.0.0:*               LISTEN      -                   
tcp6       0      0 :::22                   :::*                    LISTEN      -        

Rpi 3 model b+ / MagicMirror 2.6.0 / node -v shows v9.11.2 / Not sure how to find electron version

MichMich, 2019-02-19:
Restart your Raspberry Pi. :)

lavolp3, 2019-02-19:
Nearly everytime i Have seen that was when you tried to do `npm start dev` and the mirror was already used by pm2. 
So please also try `pm2 stop all` and then npm start dev.

closed 2019-02-26 by MichMich