MagicMirrorOrg/MagicMirror #1595: Unable to install dependencies!

ad7608327, 2019-03-02:
I am Using Raspberry Pi 3B V1.2 and automatic install and I am using automatic install and the out_put is 
root@raspberrypi:~# bash -c "$(curl -sL https://raw.githubusercontent.com/MichMich/MagicMirror/master/installers/raspberry.sh)"

$$\      $$\                     $$\           $$\      $$\ $$\                                          $$$$$$\
$$$\    $$$ |                    \__|          $$$\    $$$ |\__|                                        $$  __$$\
$$$$\  $$$$ | $$$$$$\   $$$$$$\  $$\  $$$$$$$\ $$$$\  $$$$ |$$\  $$$$$$\   $$$$$$\   $$$$$$\   $$$$$$\  \__/  $$ |
$$\$$\$$ $$ | \____$$\ $$  __$$\ $$ |$$  _____|$$\$$\$$ $$ |$$ |$$  __$$\ $$  __$$\ $$  __$$\ $$  __$$\  $$$$$$  |
$$ \$$$  $$ | $$$$$$$ |$$ /  $$ |$$ |$$ /      $$ \$$$  $$ |$$ |$$ |  \__|$$ |  \__|$$ /  $$ |$$ |  \__|$$  ____/
$$ |\$  /$$ |$$  __$$ |$$ |  $$ |$$ |$$ |      $$ |\$  /$$ |$$ |$$ |      $$ |      $$ |  $$ |$$ |      $$ |
$$ | \_/ $$ |\$$$$$$$ |\$$$$$$$ |$$ |\$$$$$$$\ $$ | \_/ $$ |$$ |$$ |      $$ |      \$$$$$$  |$$ |      $$$$$$$$\
\__|     \__| \_______| \____$$ |\__| \_______|\__|     \__|\__|\__|      \__|       \______/ \__|      \________|
                       $$\   $$ |
                       \$$$$$$  |
                        \______/

Updating packages ...
Hit:1 https://deb.nodesource.com/node_10.x stretch InRelease
Hit:2 https://dl.yarnpkg.com/debian stable InRelease
Hit:3 http://raspbian.raspberrypi.org/raspbian stretch InRelease                   
Hit:4 http://archive.raspberrypi.org/debian stretch InRelease                      
Reading package lists... Done                             
Installing helper tools ...
Reading package lists... Done
Building dependency tree       
Reading state information... Done
build-essential is already the newest version (12.3).
curl is already the newest version (7.52.1-5+deb9u9).
git is already the newest version (1:2.11.0-3+deb9u4).
unzip is already the newest version (6.0-21).
wget is already the newest version (1.18-5+deb9u2).
The following packages were automatically installed and are no longer required:
  libc-ares2 libhttp-parser2.8 libuv1 nodejs-doc realpath vlc-plugin-notify vlc-plugin-samba vlc-plugin-video-splitter vlc-plugin-visualization
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 13 not upgraded.
Check current Node installation ...
Node currently installed. Checking version number.
Minimum Node version: v5.1.0
Installed Node version: v10.15.2
No Node.js upgrade necessary.
Cloning MagicMirror ...
Cloning into 'MagicMirror'...
remote: Enumerating objects: 302, done.
remote: Counting objects: 100% (302/302), done.
remote: Compressing objects: 100% (268/268), done.
remote: Total 302 (delta 34), reused 222 (delta 24), pack-reused 0
Receiving objects: 100% (302/302), 615.65 KiB | 557.00 KiB/s, done.
Resolving deltas: 100% (34/34), done.
Cloning MagicMirror Done!
Installing dependencies ...
npm WARN deprecated time-grunt@2.0.0: Deprecated because Grunt is practically unmaintained. Move on to something better. This package will continue to work with Grunt v1, but it will not receive any updates.
npm WARN registry Unexpected warning for https://registry.npmjs.org/: Miscellaneous Warning ENETUNREACH: request to https://registry.npmjs.org/grunt failed, reason: connect ENETUNREACH 2606:4700::6810:1223:443 - Local (:::0)
npm WARN registry Using stale package data from https://registry.npmjs.org/ due to a request error during revalidation.
npm WARN deprecated nomnom@1.8.1: Package no longer supported. Contact support@npmjs.com for more info.
npm WARN tar ENOENT: no such file or directory, open '/root/MagicMirror/node_modules/.staging/normalize-package-data-7081548d/package.json'
npm WARN tar ENOENT: no such file or directory, open '/root/MagicMirror/node_modules/.staging/read-pkg-a20f90d1/package.json'
npm WARN tar ENOENT: no such file or directory, open '/root/MagicMirror/node_modules/.staging/normalize-package-data-7081548d/README.md'
npm WARN tar ENOENT: no such file or directory, open '/root/MagicMirror/node_modules/.staging/read-pkg-a20f90d1/index.js'
npm WARN tar ENOENT: no such file or directory, open '/root/MagicMirror/node_modules/.staging/normalize-package-data-7081548d/LICENSE'
npm WARN tar ENOENT: no such file or directory, open '/root/MagicMirror/node_modules/.staging/read-pkg-a20f90d1/license'
npm WARN tar ENOENT: no such file or directory, open '/root/MagicMirror/node_modules/.staging/normalize-package-data-7081548d/AUTHORS'
npm WARN tar ENOENT: no such file or directory, open '/root/MagicMirror/node_modules/.staging/read-pkg-a20f90d1/readme.md'
npm WARN tar ENOENT: no such file or directory, open '/root/MagicMirror/node_modules/.staging/jsprim-608f0a49/lib/jsprim.js'
npm WARN tar ENOENT: no such file or directory, open '/root/MagicMirror/node_modules/.staging/wdio-dot-reporter-172d5a83/lib/reporter.js'
npm WARN tar ENOENT: no such file or directory, open '/root/MagicMirror/node_modules/.staging/w3c-hr-time-b0db04bb/package.json'
npm WARN tar ENOENT: no such file or directory, open '/root/MagicMirror/node_modules/.staging/stylelint-86739683/lib/formatters/index.js'
npm WARN tar ENOENT: no such file or directory, lstat '/root/MagicMirror/node_modules/.staging/w3c-hr-time-b0db04bb/coverage/hr-time'
npm WARN tar ENOENT: no such file or directory, open '/root/MagicMirror/node_modules/.staging/stylelint-86739683/lib/formatters/jsonFormatter.js'
npm WARN tar ENOENT: no such file or directory, open '/root/MagicMirror/node_modules/.staging/stylelint-86739683/lib/formatters/needlessDisablesStringFormatter.js'
npm WARN tar ENOENT: no such file or directory, open '/root/MagicMirror/node_modules/.staging/w3c-hr-time-b0db04bb/lib/calculate-clock-offset.js'
npm WARN tar ENOENT: no such file or directory, open '/root/MagicMirror/node_modules/.staging/stylelint-86739683/lib/formatters/stringFormatter.js'
[... 587 lines cut ...]
npm WARN tar ENOENT: no such file or directory, lstat '/root/MagicMirror/node_modules/.staging/stylelint-86739683/lib/utils'
npm WARN tar ENOENT: no such file or directory, lstat '/root/MagicMirror/node_modules/.staging/stylelint-86739683/lib/utils'
npm WARN tar ENOENT: no such file or directory, lstat '/root/MagicMirror/node_modules/.staging/stylelint-86739683/lib/utils'
npm WARN tar ENOENT: no such file or directory, lstat '/root/MagicMirror/node_modules/.staging/stylelint-86739683/lib/utils'
npm WARN tar ENOENT: no such file or directory, lstat '/root/MagicMirror/node_modules/.staging/stylelint-86739683/lib/utils'
npm WARN tar ENOENT: no such file or directory, lstat '/root/MagicMirror/node_modules/.staging/stylelint-86739683/lib/utils'
npm WARN tar ENOENT: no such file or directory, lstat '/root/MagicMirror/node_modules/.staging/stylelint-86739683/lib/utils'
npm WARN tar ENOENT: no such file or directory, lstat '/root/MagicMirror/node_modules/.staging/stylelint-86739683/lib/utils'
npm WARN tar ENOENT: no such file or directory, lstat '/root/MagicMirror/node_modules/.staging/stylelint-86739683/lib/utils'
npm WARN tar ENOENT: no such file or directory, lstat '/root/MagicMirror/node_modules/.staging/stylelint-86739683/lib/utils'
npm WARN tar ENOENT: no such file or directory, lstat '/root/MagicMirror/node_modules/.staging/stylelint-86739683/lib/utils'
npm WARN tar ENOENT: no such file or directory, lstat '/root/MagicMirror/node_modules/.staging/stylelint-86739683/lib/utils'
npm WARN tar ENOENT: no such file or directory, lstat '/root/MagicMirror/node_modules/.staging/stylelint-86739683/lib/utils'
npm WARN tar ENOENT: no such file or directory, lstat '/root/MagicMirror/node_modules/.staging/stylelint-86739683/lib/utils'
npm WARN tar ENOENT: no such file or directory, lstat '/root/MagicMirror/node_modules/.staging/stylelint-86739683/lib/utils'
npm WARN tar ENOENT: no such file or directory, lstat '/root/MagicMirror/node_modules/.staging/stylelint-86739683/lib/utils'
npm WARN tar ENOENT: no such file or directory, lstat '/root/MagicMirror/node_modules/.staging/stylelint-86739683/lib/utils'
npm WARN tar ENOENT: no such file or directory, lstat '/root/MagicMirror/node_modules/.staging/stylelint-86739683/lib/utils'
npm WARN tar ENOENT: no such file or directory, lstat '/root/MagicMirror/node_modules/.staging/stylelint-86739683/lib/utils'
npm WARN tar ENOENT: no such file or directory, lstat '/root/MagicMirror/node_modules/.staging/stylelint-86739683/lib/utils'
npm WARN tar ENOENT: no such file or directory, lstat '/root/MagicMirror/node_modules/.staging/stylelint-86739683/lib/utils'
npm WARN tar ENOENT: no such file or directory, lstat '/root/MagicMirror/node_modules/.staging/stylelint-86739683/lib/utils'

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-03-02T01_13_25_900Z-debug.log

badursun, 2019-03-02:
you need to install NODE.JS first.
https://nodejs.org/en/download

If you ınstalled magıcmirror before nodejs , you should remove magicmirror app or rename folder on Pi. 
Then first install nodejs  after install magicmirror again.

I hope thats help

seramik, 2019-03-02:
i use manual install, 
node.js is installed good
but when i began " npm install && npm start", troubles appear :

ad7608327, 2019-03-02:
I already Installed it and after installing it I installed MagicMirror

terrago37, 2019-03-18:
same problem 
nodejs -v : v10.15.3
npm -v : 6.9.0

MichMich, 2019-03-18:
Please refer to http://forum.magicmirror.builders for support and troubleshooting.

I'm closing this issue for now. If you feel this issue should remain open, please let me know.

closed 2019-03-18 by MichMich

ad7608327, 2019-03-19:
@MichMich 
pls set this in setup guide
sudo apt-get install npm
sudo npm install -g npm@latest
bash -c "$(curl -sL https://raw.githubusercontent.com/MichMich/MagicMirror/master/installers/raspberry.sh)"