MagicMirrorOrg/MagicMirror #1596: No autostart of MagicMirror²

Tom-Hirschberger, 2019-03-03:
After a fresh installation of Raspbian Strech i had problems with the autostart of MagicMirror².
The Problem seems to be that the script /etc/systemd/system/pm2-root.service starts in the root environment.
I changed the lines:
USER=root to USER=pi
Environment=PM2_HOME=/root/.pm2 to Environment=PM2_HOME=/home/pi/.pm2
PIDFile=/root/.pm2/pm2.pid to PIDFile=/home/pi/.pm2/pm2.pid

Now everything works as expected.
I'm not sure if it is an Raspbian, PM2 or an MagicMirror² Issue

closed 2019-03-03 by Tom-Hirschberger

websterjazz, 2021-05-30:
To setup the Startup Script, copy/paste the following command:
`sudo env PATH=$PATH:/usr/bin /usr/lib/node_modules/pm2/bin/pm2 startup systemd -u pi --hp /home/pi`

sdetweil, 2021-05-30:
or use my script which does all the work 
see  fixuppm2.sh  (last entry on the page here)

https://github.com/sdetweil/MagicMirror_scripts

Tom-Hirschberger, 2021-05-30:
Hi,

did not have these issue in my last installations. I already use your the scripts of you @sdetweil.
That's why I closed the ticket two years ago.