MagicMirrorOrg/MagicMirror #1597: Font Awesome CSS fails to load due to incorrect MIME type.

AwesomeIndustry, 2019-03-04:
Please only submit reproducible issues. 

If you're not sure if it's a real bug or if it's just you, please open a topic on the forum: [https://forum.magicmirror.builders/category/15/bug-hunt](https://forum.magicmirror.builders/category/15/bug-hunt)
Problems installing or configuring your MagicMirror? Check out: [https://forum.magicmirror.builders/category/10/troubleshooting](https://forum.magicmirror.builders/category/10/troubleshooting)

When submitting a new issue, please supply the following information:

**Platform**: Place your platform here... give us your web browser/Electron version *and* your hardware (Raspberry Pi 2/3, Windows, Mac, Linux, System V UNIX).
Raspberry Pi 3

**Node Version**: Make sure it's version 0.12.13 or later.
v9.11.12

**MagicMirror Version**: Now that the versions have split, tell us if you are using the PHP version (v1) or the newer JavaScript version (v2).
JavaScript (v2)

**Description**: Provide a detailed description about the issue and include specific details to help us understand the problem. Adding screenshots will help describing the problem.
The Font Awesome stylesheet is not loading properly due to MIME type errors. I looked at the requests in chrome:
```
Refused to apply style from 'http://localhost:8080/vendor/node_modules/@fortawesome/fontawesome-free/css/all.min.css' because its MIME type ('text/html') is not a supported stylesheet MIME type, and strict MIME checking is enabled.
```
This also throws the same error for `localhost:8080/vendor/node_modules/@fortawesome/fontawesome_free/css/v4-shims.min.css`.
**Steps to Reproduce**: List the step by step process to reproduce the issue.
I enabled icons and fired up Chromium at `localhost:8080` to inspect the console's error messages.

**Expected Results**: Describe what you expected to see.
I expected the style sheet to load so the icons would appear.

**Actual Results**: Describe what you actually saw.
See above description.

**Configuration**: What does the used config.js file look like? Don't forget to remove any sensitive information!
<details>
<summary>Here's the calendar config portion from my `config.js` file (Click to expand)</summary>
```
		{
			module: "calendar",
			position: "middle_center",
			classes: "daytime_scheduler",
			header: "From your calendar",
			config: {
				maximumEntries: 8,
				showEnd: true,
				displaySymbol: true,
				wrapEvents: false,
				maxTitleLength: 10,
				calendars: [
					/*{
						symbol: 'calendar',
						url: "webcal://www.calendarlabs.com/templates/ical/US-Holidays.ics"
					},*/
					{
						symbol: 'calendar',
						url: "..."
					}/*,
					{
						symbol: "calendar",
						url: "..."
					},
					{
						symbol: "calendar",
						url: "..."
					}*/
				],
				titleReplace: {
					"De verjaardag van ": "",
					"'s birthday": ""
				},
				excludedEvents: [
					
				]
			}
		}
```
</details>

**Additional Notes**: Provide any other relevant notes not previously mentioned. This is optional.

Privacywonk, 2019-05-19:
Also just got hit with this after upgrading. CSS files failing to render with the following error messages in my console:

```
The resource from “http://localhost:8080/vendor/node_modules/@fortawesome/fontawesome-free/css/all.min.css” was blocked due to MIME type (“text/html”) mismatch (X-Content-Type-Options: nosniff).[Learn More] localhost:8080
The resource from “http://localhost:8080/vendor/node_modules/@fortawesome/fontawesome-free/css/v4-shims.min.css” was blocked due to MIME type (“text/html”) mismatch (X-Content-Type-Options: nosniff).[Learn More] localhost:8080
Error on loading stylesheet: vendor/css/font-awesome.css
```

Privacywonk, 2019-05-19:
Updating - went back through steps and realized I did not issue:

`npm install` after upgrading.

Once issues, this error disappeared.

closed 2019-05-21 by MichMich