MagicMirrorOrg/MagicMirror #1609: Electron not downloading its going very slow

gopigaurav, 2019-03-18:
[image]
Please only submit reproducible issues. 

If you're not sure if it's a real bug or if it's just you, please open a topic on the forum: [https://forum.magicmirror.builders/category/15/bug-hunt](https://forum.magicmirror.builders/category/15/bug-hunt)
Problems installing or configuring your MagicMirror? Check out: [https://forum.magicmirror.builders/category/10/troubleshooting](https://forum.magicmirror.builders/category/10/troubleshooting)

When submitting a new issue, please supply the following information:

**Platform**: Place your platform here... give us your web browser/Electron version *and* your hardware (Raspberry Pi 2/3, Windows, Mac, Linux, System V UNIX).

**Node Version**: Make sure it's version 0.12.13 or later.

**MagicMirror Version**: Now that the versions have split, tell us if you are using the PHP version (v1) or the newer JavaScript version (v2).

**Description**: Provide a detailed description about the issue and include specific details to help us understand the problem. Adding screenshots will help describing the problem.

**Steps to Reproduce**: List the step by step process to reproduce the issue.

**Expected Results**: Describe what you expected to see.

**Actual Results**: Describe what you actually saw.

**Configuration**: What does the used config.js file look like? Don't forget to remove any sensitive information!

**Additional Notes**: Provide any other relevant notes not previously mentioned. This is optional.

gopigaurav, 2019-03-18:
Installing dependencies then stops n shows errors

gopigaurav, 2019-03-18:
Please help

MichMich, 2019-03-18:
Please refer to http://forum.magicmirror.builders for support and troubleshooting.

I'm closing this issue for now. If you feel this issue should remain open, please let me know.

closed 2019-03-18 by MichMich

gopigaurav, 2019-03-18:
I couldnt find the solution for this problem,can u please help me out

gopigaurav, 2019-03-18:
I couldnt find the solution

MichMich, 2019-03-18:
We don't give support via GitHub. This is reserved for reproducible issues. Please **ONLY** use the forum and discord. Thanks.