MagicMirrorOrg/MagicMirror #1617: Calendars not loading

mtdjr, 2019-03-24:
**Platform**:  Docker (bastilimbach/docker-magicmirror), OSX 10.14.3

**Node Version**: `v11.12.0`

**MagicMirror Version**: Latest in repo, `latest` docker image

**Description**: This morning my calendar events were showing up while running the latest Docker image from about 3 weeks ago. Pulled in the latest Docker image and calendars stopped loading "No upcoming events". Removed my calendars and added only the default calendar (no `config` object in config.js) still no events. Says 'Loading...' for a while then no events. Tried installing on my MacBook also, same behavior. Tried wiping out all config and modules except for the calendar, same behavior. 

**Steps to Reproduce**: Run latest code. `git clone https://github.com/MichMich/MagicMirror && npm install && node serveronly`

**Expected Results**: Calendar events to show

**Actual Results**: "No upcoming events."

**Configuration**: 
```javascript
var config = {
  address: "0.0.0.0",
  port: 8080,
  ipWhitelist: [],
  language: "en",
  timeFormat: 12,
  units: "imperial",

  modules: [
    {
      module: "calendar",
      header: "Holidays",
      position: "top_left",
      config: {}
    }
  ]
};

/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== "undefined") {
  module.exports = config;
}
```

justjim1220, 2019-03-25:
It appears to be running very late...
Tracking on this end shows ETA as tomorrow. 😏😏😏

It's ok, I need to be more tolerant due to the flooding here in the
midwest. I imagine that it is the reason for taking so long. 😎🙄😎

I really appreciate your concern, I haven't dealt more than handful that
show that type of customer service.

Kudos!!!

mtdjr, 2019-03-25:
that was a weird comment. 

anyway, issue was on my end. closing

closed 2019-03-25 by mtdjr

M08IUS0NE, 2019-09-24:
> that was a weird comment.
> 
> anyway, issue was on my end. closing

What ended up being your issue/fix? I just ran into a random problem on my end with my weather, news, and time (anything network related for the most part), not loading and throwing errors in the MagicMirror start-up log that looks to be port related. Not sure yet though (haven't had time to fully dissect it but that seems to be the problem).