MagicMirrorOrg/MagicMirror #1631: Not all calendars showing after update

jsmith432, 2019-04-01:
**Platform**: Docker

**MagicMirror Version**:  (v2).

**Description**: Provide a detailed description about the issue and include specific details to help us understand the problem. Adding screenshots will help describing the problem.

**Steps to Reproduce**: 
I recently updated my magic mirror and it looks like it broke something as I am not able to see more than one calendar anymore. 

`Whoops! There was an uncaught exception...                                                                                                                                                                                                                                                                                                                                                                                                                           
TypeError: Cannot read property 'length' of undefined                                                                                                                                                                                                                                                                                                                                                                                                                
    at Object.ical.objectHandlers.END (/opt/magic_mirror/modules/default/calendar/vendor/ical.js/node-ical.js:33:22)                                                                                                                                                                                                                                                                                                                                                 
    at Object.handleObject (/opt/magic_mirror/modules/default/calendar/vendor/ical.js/ical.js:287:41)                                                                                                                                                                                                                                                                                                                                                                
    at Object.parseICS (/opt/magic_mirror/modules/default/calendar/vendor/ical.js/ical.js:328:20)                                                                                                                                                                                                                                                                                                                                                                    
    at Request._callback (/opt/magic_mirror/modules/default/calendar/vendor/ical.js/node-ical.js:11:24)                                                                                                                                                                                                                                                                                                                                                              
    at Request.self.callback (/opt/magic_mirror/modules/node_modules/request/request.js:185:22)                                                                                                                                                                                                                                                                                                                                                                      
    at Request.emit (events.js:182:13)                                                                                                                                                                                                                                                                                                                                                                                                                               
    at Request.<anonymous> (/opt/magic_mirror/modules/node_modules/request/request.js:1161:10)                                                                                                                                                                                                                                                                                                                                                                       
    at Request.emit (events.js:182:13)                                                                                                                                                                                                                                                                                                                                                                                                                               
    at Gunzip.<anonymous> (/opt/magic_mirror/modules/node_modules/request/request.js:1083:12)                                                                                                                                                                                                                                                                                                                                                                        
    at Object.onceWrapper (events.js:273:13)                                                                                                                                                                                                                                                                                                                                                                                                                         
MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?                                                                                                                                                                                                                                                                                                                                                     
If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues`

**Expected Results**: More than one calendar shown

**Actual Results**: Only one calendar is shown and no icons

**Configuration**: `		{
			module: "calendar",
			header: "Calendar",
			position: "top_right",
			config: {
				showEnd: false,
				maximumEntries: 15,
				calendars: [
					{
						symbol: "calendar-check-o ",
						url: "webcal://calendar.google.com/calendar.......",
						maximumNumberOfDays: 20
					},
					{
						symbol: "facebook-square",
						url: "webcal://www.facebook.com/events/ical/birthdays/........",
						maximumNumberOfDays: 20
					},
					{
						symbol: "birthday-cake",
						url: "webcal://calendar.google.com/calendar/ical/........",
						maximumNumberOfDays: 20
					}
				]
			}
		},`

**Additional Notes**: Provide any other relevant notes not previously mentioned. This is optional.

magic21nrw, 2019-04-03:
Same error applies here.

1|mm       | TypeError: Cannot read property 'length' of undefined
1|mm       |     at eventDate (/home/magicmirror/MagicMirror/modules/default/calendar/calendarfetcher.js:69:25)
1|mm       |     at /home/magicmirror/MagicMirror/modules/default/calendar/calendarfetcher.js:90:22
1|mm       |     at Request._callback (/home/magicmirror/MagicMirror/modules/default/calendar/vendor/ical.js/node-ical.js:11:5)
1|mm       |     at Request.self.callback (/home/magicmirror/MagicMirror/node_modules/request/request.js:185:22)
1|mm       |     at Request.emit (events.js:182:13)
1|mm       |     at Request.<anonymous> (/home/magicmirror/MagicMirror/node_modules/request/request.js:1157:10)
1|mm       |     at Request.emit (events.js:182:13)
1|mm       |     at IncomingMessage.<anonymous> (/home/magicmirror/MagicMirror/node_modules/request/request.js:1079:12)
1|mm       |     at Object.onceWrapper (events.js:273:13)
1|mm       |     at IncomingMessage.emit (events.js:187:15)

pvyParts, 2019-04-06:
i fixed this locally by updating the `calendar/vendor/ical.js` lib
https://github.com/peterbraden/ical.js

and installing the required node lib
https://www.npmjs.com/package/rrule

think it has to do with events with not enddate/time

jsmith432, 2019-04-06:
> i fixed this locally by updating the `calendar/vendor/ical.js` lib
> https://github.com/peterbraden/ical.js
> 
> and installing the required node lib
> https://www.npmjs.com/package/rrule
> 
> think it has to do with events with not enddate/time

Thank you for sharing. This fixed the events coming. I am just missing the icons now.

magic21nrw, 2019-04-06:
Hi,

this is correct. I copied the ical.js from the GitHub mentioned by @jsmith432 above into my installation and now all calendars are showing up again, the log error message is not visible anymore. 

This should be added as fix for the current repository. Anyone making a pull request?

MichMich, 2019-04-17:
Yes! Please send a PR with the tested iCal. Thanks!

MichMich, 2019-05-07:
I merged the updated ical.js in de `develop` branch. Could you guys give it a try?

Yeremiy, 2019-07-17:
Hey. I have the same problem. after the upgrade, the modules stopped working
Stopping module helper: updatenotification
Stopping module helper: calendar
Stopping module helper: newsfeed
How to fix it? thank

Yeremiy, 2019-07-17:
Hey. I have the same problem. after reinstalling the magic mirror (there was a need), the modules stopped working
Stopping module helper: updatenotification
Stopping module helper: calendar
Stopping module helper: newsfeed
How to fix it? thank

Yeremiy, 2019-07-17:
Hey. I have the same problem. after reinstalling the magic mirror (there was a need), the modules stopped working
Stopping module helper: updatenotification
Stopping module helper: calendar
Stopping module helper: newsfeed
How to fix it? I know that this is due to the latest updates. thank

Yeremiy, 2019-07-18:
reinstalling the magic mirror solved the problem. thank

closed 2019-10-11 by stale[bot]