MagicMirrorOrg/MagicMirror #1634: custom.css is being overwritten after every MM update / upgrade

evroom, 2019-04-02:
**Platform**: Raspberry Pi 3 Model B

**Node Version**: v9.11.2

**MagicMirror Version**: v2.7.1

**Description**: Every time I update or upgrade MM (using git pull && npm install), the custom.css file is being replaced with the default custom.css file (which is practically empty).

**Steps to Reproduce**: git pull && npm install

**Expected Results**: Do not replace the custom.css as it is a custom.css.

**Actual Results**: The default custom.css file.

**Configuration**: -

**Additional Notes**: Luckily I made a copy of the custom.css file.

BerHoutman, 2019-04-04:
I updated yesterday and after reboot I got a black screen. 
I didn't make a copy of  custom.css

mrvanes, 2019-04-22:
You can't track and ignore a file a the same time.
You could untrack css/custom.css, create css/custom.css.sample and ignore css/custom.css, like now and suggest users to copy custom.css.sample to custom.css to solve and keep their own. Of course, the custom.css loader logic should be robust against a missing custom.css file when users don't copy the file, or the startup script could check and copy on first start.

lavolp3, 2019-07-04:
On the update to 2.8.0 my custom.css was not overwritten.

evroom, 2019-07-05:
After the update to 2.8.0 my custom.css was overwritten, yet again.
A made a copy, so no harm done.
The main.css has been untouched 

```
pi@MagicPi:~/MagicMirror $ ll css
total 16
4 drwxr-xr-x  2 pi pi 4096 Jul  1 21:22 .
4 drwxr-xr-x 17 pi pi 4096 Jul  1 21:27 ..
4 -rw-r--r--  1 pi pi  562 Jul  1 21:22 custom.css
4 -rw-r--r--  1 pi pi 3245 Oct  5  2018 main.css
```

karanbale, 2019-07-17:
I have just experienced the same issue. After updating magic mirror using "git pull && npm install" it over-wrote my custom.css file and now I see blank screen.

sdetweil, 2019-12-29:
This should be closed, fixes in 2.9 and 2.10

closed 2019-12-29 by MichMich