MagicMirrorOrg/MagicMirror #1640: undefinded in header since update

apm1007, 2019-04-06:
Where is this undefined coming from? Seem to be there since the last update
[image]

pvyParts, 2019-04-06:
[image]
i added this line to my config and it removed it.
`appendLocationNameToHeader: false,`

apm1007, 2019-04-15:
Tried it. Isnt working for me.

Von: Aaron
Gesendet: Samstag, 6. April 2019 13:56
An: MichMich/MagicMirror
Cc: Ap Mielke; Author
Betreff: Re: [MichMich/MagicMirror] undefinded in header since update (#1640)

i added this line to my config and it removed it.
appendLocationNameToHeader: false,

MichMich, 2019-04-17:
Please refer to http://forum.magicmirror.builders for support and troubleshooting.

I'm closing this issue for now. If you feel this issue should remain open, please let me know.

closed 2019-04-17 by MichMich

victor-paumier, 2020-04-07:
Sorry to re-open this issue but I agree that there is a problem with the header.

I am not a JS expert but what I see is that as the weatherforecast do not define any header value, the value of `this.data.header` from [l.211](https://github.com/MichMich/MagicMirror/blob/447c0bffdcd2667c2d982cd2a1e5a7d43e45f112/modules/default/weatherforecast/weatherforecast.js#L211) of `weatherforecast.js` is equal to `undefined` and is concat to the `fetchedLocationName`.
To solve that, either add `header: '',` on the module definition of weatherforecast in the `config.js` file, remove `this.data.header` from this line or define a default empty value of `header` in `weatherforecast.js` (or even directly in `js/loader.js` to avoid the problem in other modules).

Anyway, this project is really great and thank you very much for that!!