MagicMirrorOrg/MagicMirror #1656: after updating 2.7.1 Magic mirror not loading.

eddy2969, 2019-04-14:
Please only submit reproducible issues. 

If you're not sure if it's a real bug or if it's just you, please open a topic on the forum: [https://forum.magicmirror.builders/category/15/bug-hunt](https://forum.magicmirror.builders/category/15/bug-hunt)
Problems installing or configuring your MagicMirror? Check out: [https://forum.magicmirror.builders/category/10/troubleshooting](https://forum.magicmirror.builders/category/10/troubleshooting)

When submitting a new issue, please supply the following information:

**Platform**: Place your platform here... give us your web browser/Electron version *and* your hardware (Raspberry Pi 2/3, Windows, Mac, Linux, System V UNIX).

**Node Version**: Make sure it's version 0.12.13 or later.

**MagicMirror Version**: Now that the versions have split, tell us if you are using the PHP version (v1) or the newer JavaScript version (v2).

**Description**: Provide a detailed description about the issue and include specific details to help us understand the problem. Adding screenshots will help describing the problem.

**Steps to Reproduce**: List the step by step process to reproduce the issue.

**Expected Results**: Describe what you expected to see.

**Actual Results**: Describe what you actually saw.

**Configuration**: What does the used config.js file look like? Don't forget to remove any sensitive information!

**Additional Notes**: Provide any other relevant notes not previously mentioned. This is optional.

eddy2969, 2019-04-14:
after updating 2.7.1 Magic mirror not loading. raspberry Pi 3 b+ just now
[image]

sdetweil, 2019-04-14:
agter installing the update with npm pull, you must run npm install, 
and also run npm install in any module folder that has a package.json file. 

some modules may need extra attention... snowboy is certainly one of them

do 
in the MMM-Hotword folder
```
npm install electron-rebuild
```
then run it
```
./node_modules/.bin/electron-rebuild
```

eddy2969, 2019-04-15:
Please.. can someone help how to fix this 
[image]

sdetweil, 2019-04-15:
those are just advisory, not blocking you...

eddy2969, 2019-04-15:
thanks for reply!  I installed magicmirror it work fine then module hotword  evething stop working it came just a black screen  what should I do?

sdetweil, 2019-04-15:
npm start dev
in the developers window, select the console tab, and scroll up for errors (red text)..

did u work thru the MMM-Hotword install 'in case of errors'...

eddy2969, 2019-04-15:
Thank you very much. I really appreciate  for your help I will  start all over again will let you know I need your brain for this I'm very new. Thank you!

kumarsatish23, 2019-04-15:
Even I faced the same issue after updating mm2 just you have to re install Hotword module again follow instruction provided in https://github.com/eouia/MMM-Hotword.
In my case 
`cd ~/MagicMirror/modules/MMM-Hotword/node_modules/snowboy
npm install --save-dev electron-rebuild
npm install nan
./node_modules/.bin/electron-rebuild `
these command worked for me.

MichMich, 2019-04-17:
Disable all non default modules. Most of the time those are the reason.

closed 2019-05-07 by MichMich