MagicMirrorOrg/MagicMirror #1664: Power save with no signal?

dxfan227, 2019-04-21:
So this is kind of related but I could t find an answer anywhere so I am hoping someone has an idea here.

I am using mmm motion detector to turn my monitor on and off.. it works well but after a prolonged “off” /“no signal” the motion won’t turn it in anymore. I can’t even get the screen to turn on with keyboard and mouse.

The pi is still powered on and I can ssh and ping it. I can’t figure out what’s wrong.

closed 2019-05-03 by dxfan227

JoshuaShelswell, 2020-12-20:
did you ever figure this out? im noticing the same issue

rejas, 2020-12-20:
Which motion detector module are you using?

JoshuaShelswell, 2020-12-20:
> 
> 
> Which motion detector module are you using?

yours actually, been trying to work out the bugs before i present the present to my partners mom lol

JoshuaShelswell, 2020-12-20:
I uncommented

hdmi_force_hotplug=1
config_hdmi_boost=4

and added

hdmi_blanking=0

not sure if that resolved the issue but it hasnt done it in the last hour, the screensaver is also disabled

JoshuaShelswell, 2020-12-21:
the added lines seems to have fixed it for now, motion detecting now acting as it should