MagicMirrorOrg/MagicMirror #1667: Type Error Main.js line 119 and 139

dxfan227, 2019-04-26:
Please only submit reproducible issues. 

If you're not sure if it's a real bug or if it's just you, please open a topic on the forum: [https://forum.magicmirror.builders/category/15/bug-hunt](https://forum.magicmirror.builders/category/15/bug-hunt)
Problems installing or configuring your MagicMirror? Check out: [https://forum.magicmirror.builders/category/10/troubleshooting](https://forum.magicmirror.builders/category/10/troubleshooting)

When submitting a new issue, please supply the following information:

**Platform**: Raspberry Pi 3B

**Node Version**: Not sure

**MagicMirror Version**: Not sure

**Description**:  When launching the magic mirror via nom start dev. In the console I just keep getting the following error
Type ErrorL Cannot read properly
'get ellementbyclassname' of null
Main.js LINE 139 and 119

what is this?

sdetweil, 2019-04-26:
some module is returning NULL from its getDom()  function.. the full statcktrace should show the failing module

dxfan227, 2019-04-26:
can you explain how to " show the full stacktrace" so I can pin this down? im very new, thank you for your help.

sdetweil, 2019-04-26:
if you open the developers window, ctrl-shift-i, or start MM with npm start dev, then select the console tab... you should find the error there..

dxfan227, 2019-04-26:
Hmm, that's where i saw the original error message..in the console LOL. I'll look again and see ifI can find out any more info.

I'm actually looking for a reason my motion detector module keeps turning off after 10-15 min. Is there anything you can think of that would cause the Raspberry pi/electron/magic mirror to make a module like mmm-motiondetecor to turn off my webcam after 10-15 min of working normally?

sdetweil, 2019-04-26:
looking thru the diff cam engine code and the motion detector project I don't see any place where the camera is turned off. i use a web cam too.. but use the motion project  for camera mgmt...

dxfan227, 2019-04-26:
This is why it is so strange.... the developer of that module also said that his module does not control the camera.

I can’t understand why mine keeps turning off after 10-15 min. The camera is still working as it is powered on and the microphone still is active for hotword/assistant.

When magic mirror first launches, it works very well and is active with a red light to indicate capturing video.... and reacts to motions... but after it deactivated the monitor... and I wait 10-15 min the red light turns off and the camera no longer turns the monitor on or off.

 I’m pulling my hair out( what’s left of it)

fewieden, 2019-04-28:
@dxfan227 do you might have a screensaver activated. what happens if zou move the mouse after it is not reacting anymore?

dxfan227, 2019-04-28:
Thanks for the reply. So when I move the mouse nothing happens... the machine required me to manually send the activate monitor command in order for the screen to turn back on as the module turned off and then the camera turned itself off.

I ended up installing a different module called motion-detection..... this one works and does not time out or anything. 

In the end it works out but I still wish I knew what was wrong befor

closed 2019-05-07 by MichMich