MagicMirrorOrg/MagicMirror #1668: Automatic Start & Doesn't Shut Down

randomrushgirl, 2019-04-27:
Hi I'm a noob and I just started with Raspberry Pi projects. So this is my first major project and I want to be able to customise later on but for now I'm following everything word by word.
So I ran the following command as specified in README.md after the initial configurations:
`bash -c "$(curl -sL https://raw.githubusercontent.com/MichMich/MagicMirror/master/installers/raspberry.sh)"`
After the automatic installation took place, I was reading about the next steps when lo and behold, without prompting the application to run, it starts on its own and when I pressed Ctrl+Q, after a few seconds, it restarted again without any prompting! Please someone help! I can't get it to stop.

sdetweil, 2019-04-27:
that is the pm2 automatic control... 

do pm2 list to see the applications being run, and then pm2 stop the application name (I think it is 'MagicMirror')

randomrushgirl, 2019-04-28:
Oh thank you very much :)
So if I wanted to run Alexa like this simultaneously with the MagicMirror I’d have to do a pm for that in a similar way?

sdetweil, 2019-04-28:
There is a module to run Alexa inside mm.
MMM-Alexa on github

Marcreed48, 2019-04-28:
This may help.https://github.com/MichMich/MagicMirror/wiki/Voice-Related-Modules

dxfan227, 2019-05-03:
My pm2 is pm2 stop mm

randomrushgirl, 2019-05-05:
Thanks sdetweil, dxfan227, Marcreed48 💯 you've been incredibly helpful

closed 2019-05-07 by MichMich