MagicMirrorOrg/MagicMirror #1672: Audio input and output

smartmirrorcse, 2019-05-04:
Please only submit reproducible issues. 

If you're not sure if it's a real bug or if it's just you, please open a topic on the forum: [https://forum.magicmirror.builders/category/15/bug-hunt](https://forum.magicmirror.builders/category/15/bug-hunt)
Problems installing or configuring your MagicMirror? Check out: [https://forum.magicmirror.builders/category/10/troubleshooting](https://forum.magicmirror.builders/category/10/troubleshooting)

When submitting a new issue, please supply the following information:

**Platform**: Place your platform here... give us your web browser/Electron version *and* your hardware (Raspberry Pi 2/3, Windows, Mac, Linux, System V UNIX).

**Node Version**: Make sure it's version 0.12.13 or later.

**MagicMirror Version**: Now that the versions have split, tell us if you are using the PHP version (v1) or the newer JavaScript version (v2).

**Description**: Provide a detailed description about the issue and include specific details to help us understand the problem. Adding screenshots will help describing the problem.

**Steps to Reproduce**: List the step by step process to reproduce the issue.

**Expected Results**: Describe what you expected to see.

**Actual Results**: Describe what you actually saw.

**Configuration**: What does the used config.js file look like? Don't forget to remove any sensitive information!

**Additional Notes**: Provide any other relevant notes not previously mentioned. This is optional.

smartmirrorcse, 2019-05-04:
Hi . I am almost done with my Smart mirror project but I have issue with the audio input and output.
I'am using ras pi 3 onboard 3.5mm jack for audio output and usb microphone  as audio in.
both are working individually  absolutely fine  but when I record something using usb microphone I am unable to get the output on the 3.5 mm microphones.
so I would really appreciate some help  as soon as possible.I have a demonstration of the project  within few days.

sdetweil, 2019-05-04:
sorry, don't understand... 
```
 I am unable to get the output on the 3.5 mm **microphones**
```
do you mean speakers?

if you use
 ```
arecord somefile.wav 
```
to record from the usb mic
then 
```
aplay somefile.wav
could play to the output device (called default)..  

do you get system sounds from the analog device port? 

note the cable matters and powered speakers
https://www.raspberrypi-spy.co.uk/2014/07/raspberry-pi-model-b-3-5mm-audiovideo-jack/

and look at raspi-config
https://raspberrypi.stackexchange.com/questions/61298/force-sound-over-3-5mm-jack-and-video-over-hdmi/61304#61304

smartmirrorcse, 2019-05-04:
By 3.5mm i mean earphones which can be plugged on to 3.5mm port of ras pi 3.
Here's my .asoundrc file :-
cm.!default{
  type asym
  playback.pcm{
    type hw
    card 0
  }
  capture.pcm{
    type plug
    slave.pcm "hw:1, 0"
  }
}

ctl.!default{
  type hw
  card 0
}
The command for audio input through usb microphone is :-
1.For recording
pi@raspberrypi:~ $ arecord --format=S16_LE --duration=5 --rate=16000 --file-type=raw out.raw
2.For playing the recorded data
pi@raspberrypi:~ $ aplay --format=S16_LE --rate=16000 out.raw

so now my issue is when I play the recorded data there is no output in earphones which is connected to 3.5 mm jack of ras pi 3.
But when I play youtube in pi I am able to get the sound output in earphones.

sdetweil, 2019-05-04:
do aplay -l (lowercase L) to list the output devices..   make sure your .asoundrc file is right

should be able to specify device too on aplay commandline -D xx:yy

smartmirrorcse, 2019-05-04:
sudo aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 7/7
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

sdetweil, 2019-05-04:
see this topic.. links to PI config
https://forum.magicmirror.builders/topic/10285/no-sound-from-my-rpi-3-b?page=1

smartmirrorcse, 2019-05-05:
Hi could I get your email please.
Because nothing is working out for me.

smartmirrorcse, 2019-05-05:
HI I have started over with new Os.
pi@raspberrypi:~/MagicMirror/modules/MMM-Hotword $ npm install
npm WARN deprecated hawk@3.1.3: This module moved to @hapi/hawk. Please make sure to switch over as this distribution is no longer supported and may contain bugs and critical security issues.
npm WARN deprecated hoek@2.16.3: This version has been deprecated in accordance with the hapi support policy (hapi.im/support). Please upgrade to the latest version to get the best features, bug fixes, and security patches. If you are unable to upgrade at this time, paid support is available for older versions (hapi.im/commercial).
npm WARN deprecated boom@2.10.1: This version has been deprecated in accordance with the hapi support policy (hapi.im/support). Please upgrade to the latest version to get the best features, bug fixes, and security patches. If you are unable to upgrade at this time, paid support is available for older versions (hapi.im/commercial).
npm WARN deprecated cryptiles@2.0.5: This version has been deprecated in accordance with the hapi support policy (hapi.im/support). Please upgrade to the latest version to get the best features, bug fixes, and security patches. If you are unable to upgrade at this time, paid support is available for older versions (hapi.im/commercial).
npm WARN deprecated sntp@1.0.9: This module moved to @hapi/sntp. Please make sure to switch over as this distribution is no longer supported and may contain bugs and critical security issues.

> snowboy@1.3.1 install /home/pi/MagicMirror/modules/MMM-Hotword/node_modules/snowboy
> node-pre-gyp install --fallback-to-build

[snowboy] Success: "/home/pi/MagicMirror/modules/MMM-Hotword/node_modules/snowboy/lib/node/binding/Release/node-v57-linux-arm/snowboy.node" is installed via remote
npm notice created a lockfile as package-lock.json. You should commit this file.
npm WARN MMM-Hotword@1.1.0 No repository field.

added 216 packages from 133 contributors and audited 565 packages in 213.194s
found 11 vulnerabilities (8 moderate, 3 high)
  run `npm audit fix` to fix them, or `npm audit` for details

Is this ok can I proceed.

sdetweil, 2019-05-05:
come find me on discord and I will give you my email there..

as for hotword, it depends, if u start MagicMirror does it work or fail?

smartmirrorcse, 2019-05-05:
sure I will. 
well I had a other query as well .
pi@raspberrypi:~/MagicMirror/modules/MMM-AssistantMk2 $ ./node_modules/.bin/electron-rebuild
⠧ Building module: grpc, Completed: 0

This build is taking quite long.

sdetweil, 2019-05-05:
Yes. One took 30 minutes for me

smartmirrorcse, 2019-05-05:
so should I wait for it to complete.

sdetweil, 2019-05-05:
Yes

closed 2019-05-07 by MichMich