MagicMirrorOrg/MagicMirror #1676: Location ID link doesn't work for weather (openweathermap) - I found a link that has a JSON file with the location IDs. Follow this link - http://78.46.48.103/sample/

johnbosak, 2019-05-14:
Please only submit reproducible issues. 

If you're not sure if it's a real bug or if it's just you, please open a topic on the forum: [https://forum.magicmirror.builders/category/15/bug-hunt](https://forum.magicmirror.builders/category/15/bug-hunt)
Problems installing or configuring your MagicMirror? Check out: [https://forum.magicmirror.builders/category/10/troubleshooting](https://forum.magicmirror.builders/category/10/troubleshooting)

When submitting a new issue, please supply the following information:

**Platform**: Place your platform here... give us your web browser/Electron version *and* your hardware (Raspberry Pi 2/3, Windows, Mac, Linux, System V UNIX).

**Node Version**: Make sure it's version 0.12.13 or later.

**MagicMirror Version**: Now that the versions have split, tell us if you are using the PHP version (v1) or the newer JavaScript version (v2).

**Description**: Provide a detailed description about the issue and include specific details to help us understand the problem. Adding screenshots will help describing the problem.

**Steps to Reproduce**: List the step by step process to reproduce the issue.

**Expected Results**: Describe what you expected to see.

**Actual Results**: Describe what you actually saw.

**Configuration**: What does the used config.js file look like? Don't forget to remove any sensitive information!

**Additional Notes**: Provide any other relevant notes not previously mentioned. This is optional.

vincep5, 2019-05-14:
Can you show that portion of your config file?  The locationID should be in double quotes.

johnbosak, 2019-05-14:
Thanks for your quick response. This is not a configuration file problem.  The issue has to do with finding the Location ID for the configuration file.  The "read me" file says:locationID: "", //Location ID from http://openweathermap.org/help/city_list.txtThis link does not work.  In my case, specifying the location does not get me the correct city.  My city is Cornelius, NC.  So I put this in the config file: location: "cornelius, US"Because there is more than one Cornelius in the US the one I get ( the first one in the list) is not the correct one.  Therefore, I must use the Location ID to specify the correct location.  I found this location list by downloading the "city.list.json.gz" file at :
http://78.46.48.103/sample/

I hope this make sense.

vincep5, 2019-05-14:
Ah, okay.  Yes, the link is should be replaced with http://bulk.openweathermap.org/sample/city.list.json.gz in all areas that reference http://openweathermap.org/help/city_list.txt in the code comments.

Just checking you were in fact able to find your town's proper ID yes ?

johnbosak, 2019-05-14:
Yes. 

John Bosak

MichMich, 2019-05-17:
Merged.

closed 2019-05-17 by MichMich