MagicMirrorOrg/MagicMirror #1679: Black Screen after Desktop setup

djay07, 2019-05-15:
Good morning,
i have a black screen since yesterday evening.
I read all "black screen" threads i found, without success.

**What I did:**
- Install full rasbian
- run install script
- play with some modules (all good till here, for two evenings)
- adjusted rasbian desktop settings (to get rid of the window pop-ups (locale changed + changed the background to black)
- since this moment i have a black screen.

I tried reverting everything i changed, without success.
The installation is from 3 days ago, so everything should be up2date.
As said, the config.js was working fine until i tuned the rasbian desktop, to be more "invisible on startup". So i don't expect any issue there.
Also the config checker says: all cool.

**The log shows me:**

```
/home/pi/.pm2/logs/MagicMirror-error.log last 15 lines:
0|MagicMir | ATTENTION: default value of option force_s3tc_enable overridden by environment.
0|MagicMir |
0|MagicMir | (process:1237): Gtk-WARNING **: Locale not supported by C library.
0|MagicMir |    Using the fallback 'C' locale.
0|MagicMir | libGL error: MESA-LOADER: failed to retrieve device information
0|MagicMir | MESA-LOADER: failed to retrieve device information
0|MagicMir | MESA-LOADER: failed to retrieve device information
0|MagicMir | ATTENTION: default value of option force_s3tc_enable overridden by environment.
0|MagicMir |
0|MagicMir | (process:1337): Gtk-WARNING **: Locale not supported by C library.
0|MagicMir |    Using the fallback 'C' locale.
0|MagicMir | libGL error: MESA-LOADER: failed to retrieve device information
0|MagicMir | MESA-LOADER: failed to retrieve device information
0|MagicMir | MESA-LOADER: failed to retrieve device information
0|MagicMir | ATTENTION: default value of option force_s3tc_enable overridden by environment.
```

```
> magicmirror@2.7.1 config:check /home/pi/MagicMirror
> node tests/configs/check_config.js

Checking file...  /home/pi/MagicMirror/config/config.js
Your configuration file doesn't contain syntax errors :)

```

```
^[[Api@raspberrypi:~ $ pm2 info MagicMirror
 Describing process with id 0 - name MagicMirror
┌───────────────────┬──────────────────────────────────────────┐
│ status            │ online                                   │
│ name              │ MagicMirror                              │
│ version           │ 2.7.1                                    │
│ restarts          │ 0                                        │
│ uptime            │ 95s                                      │
│ script path       │ /home/pi/MagicMirror/installers/mm.sh    │
│ script args       │ N/A                                      │
│ error log path    │ /home/pi/.pm2/logs/MagicMirror-error.log │
│ out log path      │ /home/pi/.pm2/logs/MagicMirror-out.log   │
│ pid path          │ /home/pi/.pm2/pids/MagicMirror-0.pid     │
│ interpreter       │ bash                                     │
│ interpreter args  │ N/A                                      │
│ script id         │ 0                                        │
│ exec cwd          │ /home/pi/MagicMirror                     │
│ exec mode         │ fork_mode                                │
│ node.js version   │ N/A                                      │
│ node env          │ N/A                                      │
│ watch & reload    │ ✔                                        │
│ unstable restarts │ 0                                        │
│ created at        │ 2019-05-13T18:37:24.437Z                 │
└───────────────────┴──────────────────────────────────────────┘
 Revision control metadata
┌──────────────────┬──────────────────────────────────────────────────┐
│ revision control │ git                                              │
│ remote url       │ https://github.com/MichMich/MagicMirror.git      │
│ repository root  │ /home/pi/MagicMirror                             │
│ last update      │ 2019-05-15T06:37:32.480Z                         │
│ revision         │ b508a629e8a727358f1b88e3c7df842d14669ec4         │
│ comment          │ Merge pull request #1632 from MichMich/fix-2.7.1 │
│                  │                                                  │
│                  │ Fix package.json version number.                 │
│ branch           │ master                                           │
└──────────────────┴──────────────────────────────────────────────────┘
 Divergent env variables from local env
┌────────────────┬─────────────────────────────────────────┐
│ LC_ALL         │ N/A                                     │
│ SSH_CONNECTION │ 192.168.xxx.28 51340 192.168.xxx.212 22 │
│ LANG           │ en_GB.UTF-8                             │
│ PWD            │ /home/pi/MagicMirror                    │
│ SSH_CLIENT     │ 192.168.xxx.28 51340 22                 │
│ SHLVL          │ 2                                       │
│ LANGUAGE       │ N/A                                     │
└────────────────┴─────────────────────────────────────────┘

```

**I appreciate any advise.**

sdetweil, 2019-05-15:
but what is black?  the whole system? boot and not MagicMirror? if u just changed system things, MM would not be affected..

if MM black screen, this means a syntax error in some modules node_helper more than likely...

what is in he pm2 logs?
 error log path    │ /home/pi/.pm2/logs/MagicMirror-error.log │
│ out log path      │ /home/pi/.pm2/logs/MagicMirror-out.log   │

djay07, 2019-05-15:
Well when the RPi boots, it shows the desktop for a few secs, like it should and when it switches to MM it goes black.

This is all i changed, since i usually go step by step to revert if needed. 
Anyway, the logs:

```
  GNU nano 2.7.4                                   File: /home/pi/.pm2/logs/MagicMirror-error.log

ATTENTION: default value of option force_s3tc_enable overridden by environment.
libGL error: MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
ATTENTION: default value of option force_s3tc_enable overridden by environment.
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! magicmirror@2.7.1 start: `sh run-start.sh`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the magicmirror@2.7.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2019-05-13T19_02_12_594Z-debug.log
libGL error: MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
ATTENTION: default value of option force_s3tc_enable overridden by environment.
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! magicmirror@2.7.1 start: `sh run-start.sh`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the magicmirror@2.7.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2019-05-13T19_05_28_209Z-debug.log
libGL error: MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
ATTENTION: default value of option force_s3tc_enable overridden by environment.
WARNING! Could not validate config file. Starting with default configuration. Please correct syntax errors at or above this$
                        module: "clock",
                              ^

SyntaxError: Unexpected token :
    at new Script (vm.js:74:7)

```

```
GNU nano 2.7.4                            File: /home/pi/.npm/_logs/2019-05-15T06_27_42_222Z-debug.log

0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@6.4.1
3 info using node@v10.15.3
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle magicmirror@2.7.1~prestart: magicmirror@2.7.1
6 info lifecycle magicmirror@2.7.1~start: magicmirror@2.7.1
7 verbose lifecycle magicmirror@2.7.1~start: unsafe-perm in lifecycle true
8 verbose lifecycle magicmirror@2.7.1~start: PATH: /usr/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/home/$
9 verbose lifecycle magicmirror@2.7.1~start: CWD: /home/pi/MagicMirror
10 silly lifecycle magicmirror@2.7.1~start: Args: [ '-c', 'sh run-start.sh' ]
11 silly lifecycle magicmirror@2.7.1~start: Returned: code: 1  signal: null
12 info lifecycle magicmirror@2.7.1~start: Failed to exec start script
13 verbose stack Error: magicmirror@2.7.1 start: `sh run-start.sh`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:301:16)
13 verbose stack     at EventEmitter.emit (events.js:189:13)
13 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at ChildProcess.emit (events.js:189:13)
13 verbose stack     at maybeClose (internal/child_process.js:970:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:259:5)
14 verbose pkgid magicmirror@2.7.1
15 verbose cwd /home/pi/MagicMirror
16 verbose Linux 4.14.98-v7+
17 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"
18 verbose node v10.15.3
19 verbose npm  v6.4.1
20 error code ELIFECYCLE
21 error errno 1
22 error magicmirror@2.7.1 start: `sh run-start.sh`
22 error Exit status 1
23 error Failed at the magicmirror@2.7.1 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]

```

config.js snibbet:

```
        modules: [
                {
                        module: "alert",
                },
                {
                        module: "updatenotification",
                        position: "top",
                },
                {
                        module: "clock",
                        header: "Gießen",
                        position: "top center",   // This can be any of the regions.
                        config: {
                                timeFormat: 24,
                                timezone: "Europe/Berlin",
                        }
                },
```

i deleted the folders of unused modules and re-run the "npm install" cmd in the MM root folder.
Now i see a syntax error i didn't see before in the logs. I am a bit lost here.

sdetweil, 2019-05-15:
did u have a running system and upgraded to 2.7.1?

if so, did u check and do 
````
npm install
```
in any module folder that has a package.,json file?

u must refresh any modules dependencies to match the base mirror level code

djay07, 2019-05-15:
> did u have a running system and upgraded to 2.7.1?

No, I freshly installed it on Monday evening, from scratch.
_Rasbian > update + upgrade > MM automatic install cmd_

> in any module folder that has a package.,json file?
- in the modules folder itself there is a package-lock.json
- the MMM-GoogleMapsTraffic module has a package.json + package-lock.json file.

> u must refresh any modules dependencies to match the base mirror level code
how can i do that?

> > > > Now i see a syntax error i didn't see before in the logs. I am a bit lost here.
> 
> what syntax error? i don't see it in the info you provided

this:
```
ATTENTION: default value of option force_s3tc_enable overridden by environment.
WARNING! Could not validate config file. Starting with default configuration. Please correct syntax errors at or above this$
                        module: "clock",
                              ^

SyntaxError: Unexpected token :
    at new Script (vm.js:74:7)
```

sdetweil, 2019-05-15:
ok, need  to see config.js

MichMich, 2019-05-15:
Black screens are usually due to:
- 3rd party modules
- errors in config file

Also, please use the forum for support. These GitHub issues are meant for reproducible issues. Not for support.

Please refer to http://forum.magicmirror.builders for support and troubleshooting.

I'm closing this issue for now. If you feel this issue should remain open, please let me know.

closed 2019-05-15 by MichMich

djay07, 2019-05-15:
> ok, need to see config.js

just tested here:
https://jshint.com/
as mentioned in the wiki and there are no issues reported.

> Also, please use the forum for support. These GitHub issues are meant for reproducible issues. Not for support.

I realized that after posting. Sorry for that. Will start from scratch and post in the forum if needed.

djay07, 2019-05-16:
FYI:
i did a new setup on a second sd card yesterday evening and the config.js file works like a charm.
I hope the hole thing doesn't break again "just like that".