MagicMirrorOrg/MagicMirror #1681: Pi3-- MM Does not start after fresh install of it.

pancakeykakes, 2019-05-18:
Used both curl and manual install methods.
There are warnings of a few packages being deprecated that show up during install.
'This is probably not a problem with npm' is a bold statement for a freshly installed program to make, by the way. Kind of off-putting. Might I suggest something a bit less brash like "hey check the log and see what it says"?

Anyway this is the log generated whenever trying to run a fresh install of MM on my RPi3b:

0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@6.4.1
3 info using node@v10.15.3
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle magicmirror@2.7.1~prestart: magicmirror@2.7.1
6 info lifecycle magicmirror@2.7.1~start: magicmirror@2.7.1
7 verbose lifecycle magicmirror@2.7.1~start: unsafe-perm in lifecycle true
8 verbose lifecycle magicmirror@2.7.1~start: PATH: /usr/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/home/pi/MagicMirror/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/$
9 verbose lifecycle magicmirror@2.7.1~start: CWD: /home/pi/MagicMirror
10 silly lifecycle magicmirror@2.7.1~start: Args: [ '-c', 'sh run-start.sh' ]
11 silly lifecycle magicmirror@2.7.1~start: Returned: code: 1  signal: null
12 info lifecycle magicmirror@2.7.1~start: Failed to exec start script
13 verbose stack Error: magicmirror@2.7.1 start: `sh run-start.sh`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:301:16)
13 verbose stack     at EventEmitter.emit (events.js:189:13)
13 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at ChildProcess.emit (events.js:189:13)
13 verbose stack     at maybeClose (internal/child_process.js:970:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:259:5)
14 verbose pkgid magicmirror@2.7.1
15 verbose cwd /home/pi/MagicMirror
16 verbose Linux 4.9.35-v7+
17 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"
18 verbose node v10.15.3
19 verbose npm  v6.4.1
20 error code ELIFECYCLE
21 error errno 1
22 error magicmirror@2.7.1 start: `sh run-start.sh`
22 error Exit status 1
23 error Failed at the magicmirror@2.7.1 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]

Marcreed48, 2019-05-18:
Happened to me also, it worked after I flashed a fresh raspbian stretch.

MichMich, 2019-05-18:
Please refer to http://forum.magicmirror.builders for support and troubleshooting.

I'm closing this issue for now. If you feel this issue should remain open, please let me know.

closed 2019-05-18 by MichMich

pancakeykakes, 2019-05-18:
Just made a fresh install of stretch on raspi3b, then installed MM. Same results.
`0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@6.4.1
3 info using node@v10.15.3
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle magicmirror@2.7.1~prestart: magicmirror@2.7.1
6 info lifecycle magicmirror@2.7.1~start: magicmirror@2.7.1
7 verbose lifecycle magicmirror@2.7.1~start: unsafe-perm in lifecycle true
8 verbose lifecycle magicmirror@2.7.1~start: PATH: /usr/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/home/pi/MagicMirror/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/$
9 verbose lifecycle magicmirror@2.7.1~start: CWD: /home/pi/MagicMirror
10 silly lifecycle magicmirror@2.7.1~start: Args: [ '-c', 'sh run-start.sh' ]
11 info lifecycle magicmirror@2.7.1~start: Failed to exec start script
12 verbose stack Error: magicmirror@2.7.1 start: `sh run-start.sh`
12 verbose stack spawn ENOENT
12 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:48:18)
12 verbose stack     at ChildProcess.emit (events.js:189:13)
12 verbose stack     at maybeClose (internal/child_process.js:970:16)
12 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:259:5)
13 verbose pkgid magicmirror@2.7.1
14 verbose cwd /home/pi/MagicMirror
15 verbose Linux 4.14.98-v7+
16 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"
17 verbose node v10.15.3
18 verbose npm  v6.4.1
19 error file sh
20 error code ELIFECYCLE
21 error errno ENOENT
22 error syscall spawn
23 error magicmirror@2.7.1 start: `sh run-start.sh`
23 error spawn ENOENT
24 error Failed at the magicmirror@2.7.1 start script.
24 error This is probably not a problem with npm. There is likely additional logging output above.
25 verbose exit [ 1, true ]

`

Please consider re-opening.