MagicMirrorOrg/MagicMirror #1684: Why currentweather module is using lockString?

eouia, 2019-05-21:
It is unable to hide default `currentweather` and `weatherforecast` modules, because those modules are using `.show()` and `.hide()` **with `lockString`** to beautify its' appearance. So other module cannot hide them easily. 
Is there any reason or profit ? I cannot figure out.

MichMich, 2019-05-21:
A lock string does not prevent an other model from hiding it. It only prevents it from showing. 

Please refer to http://forum.magicmirror.builders for support and troubleshooting.

I'm closing this issue for now. If you feel this issue should remain open, please let me know.

closed 2019-05-21 by MichMich

eouia, 2019-05-21:
You are right. Just using `Othermodule.hide({lockString:"something")` could work. Sorry for bothering you.

MichMich, 2019-05-21:
No worries! Enjoy developing!