MagicMirrorOrg/MagicMirror #1686: RSS wrapDescription doesnt work

schlittrix, 2019-05-21:
Hey guys,
i am new in this Business but I got a Problem with the RSS Feed function called wrapDescription. 
I activate the Description below the headline, but the Option wrapDescription doesnt work anymore. 
Is there a solution for it?

MichMich, 2019-05-22:
Please refer to http://forum.magicmirror.builders for support and troubleshooting.

I'm closing this issue for now. If you feel this issue should remain open, please let me know.

closed 2019-05-22 by MichMich

eric29, 2019-07-14:
That was the laziest response from the Mich. The property is BROKEN and is an "issue".

Here's a quick "sorta" fix for now. In the "MagicMirror/css/custom.css" file, add these lines:

`.newsfeed-desc {
	width: 400px;  /* Change to your own width if you want */
}`

It won't "wrap" the text, but at least lets you control the width of the container with the description in it.

The issue I believe is that it sizes the "div" container first to the size of the "un-wrapped" description, then wraps the text. So essentially it doesn't get wrapped really. I think... Either way, it doesn't work.