MagicMirrorOrg/MagicMirror #1687: Display emoji

tisnatch, 2019-05-24:
As an example, Google Calendar allows you to display emoji in event titles. The emoji are not imported correctly into the MM interface. Would it be possible to correctly show emoji? 

**Platform**: Electron 5.0.2 / Raspberry Pi 3 B+ / Stretch.

**Node Version**: 10.15.3

**MagicMirror Version**:  JavaScript version (v2).

**Description**: Emoji not being displayed.

**Configuration**: Nothing special; calendar displays correctly except for emoji

MichMich, 2019-05-24:
Not sure if the Raspberry Pi Desktop environment is able to display emoji's. If so, it might be an encoding issue.

tisnatch, 2019-05-24:
I'll see if I can solve it by installing different emoji centric fonts in
the OS. Do you know if Electron depends on the OS to render fonts?

MichMich, 2019-05-24:
I think so. Yes.

ASSYANA, 2019-05-24:
You can label columns with status indicators like,

MichMich, 2019-06-14:
Closed due to inactivity.

closed 2019-06-14 by MichMich