MagicMirrorOrg/MagicMirror #1696: Auth to calendar not working

Hg347, 2019-06-09:
**Description**: 
I think the authentication, error logging or event system respectively is not working properly.
If you try to auth against a calender like this:
https://5a15a-de.netcup-mail.de/SOGo/dav/raspi@5a15a.de/Calendar/personal.ics

There is not error message, none of the console.log() messages or anything. It simply does not show up.
If you put above link in your browser, a login screen is presented. In case you provide the correct credentials, the ics file can be accessed.

**Steps to Reproduce**: 
try to show https://5a15a-de.netcup-mail.de/SOGo/dav/raspi@5a15a.de/Calendar/personal.ics
in your calendar module.

**Expected Results**: 
At least auth errors or messages in browser console.

**Actual Results**: 
nothing of above
[image]

**Configuration**: 

```
modules: [
                {
                        module: "calendar",
                        position: "top_left",
                        config: {
                                colored: true,
                                coloredSymbolOnly: true,
                                calendars: [
                                        {
                                                url: "https://5a15a-de.netcup-mail.de/SOGo/dav/raspi@5a15a.de/Calendar/personal.ics",
                                                color: "#d5b2ff",
                                                auth: {
                                                        user: "rsdfe",
                                                        pass: ".dfsaas",
                                                        method: "basic"
                                                }
                                        },
                                        {
                                                symbol: "couch",
                                                url: "https://www.thunderbird.net/media/caldata/GermanHolidays.ics",
                                                color: "#c2ff89"
                                        }

                                ]
                        }
                },

        ]
```

**Additional Notes**: 
The calendar to access is of SOGo Groupware.

sdetweil, 2019-06-09:
the code in the fetcher (gets the ics file) says 
```
if (auth) {
			if(auth.method === "bearer"){
				opts.auth = {
					bearer: auth.pass
				}

			}else{
				opts.auth = {
					user: auth.user,
					pass: auth.pass
				};

				if(auth.method === "digest"){
					opts.auth.sendImmediately = false;
				}else{
					opts.auth.sendImmediately = true;
				}
			}
		}
}
```
i don't know what 'sendImmediatrely' means, but maybe u want to try  method: "digest"

as u noted there are no error messages for failed authentication

Hg347, 2019-06-09:
same result with "digest" setting.

Hg347, 2019-06-09:
Well, I am not familiar with JS or debugging JS. But the browser debug feature seems to show, that several files are not involved in fetching the ics files?
How can I set a breakpoint in calendarfetcher.js? It is not even listed in the browser.

[image]

sdetweil, 2019-06-09:
fetcher is run in the background , so you cannot interactively debug in the UI side of the mirror app.

u can add console.log stmts to the fetcher code and see them when u start the mirror like this
```
npm start &
```
they will show in the terminal window

I use ssh to connect from my PC to the mirror so I can do all this debugging..

you could add code here , line 57 of fetcher
```
			if (err) {
				fetchFailedCallback(self, err);
```
to output the error..  my 'guess' is that this site wanted something other and user/pass tokens

the fetchFailedCallback is just an empty routine currently

Hg347, 2019-06-09:
That makes sense. So some code is run on server side only and some on client side. Thank you for the info.

Hg347, 2019-06-09:
Got the error at least:
```
 Whoops! There was an uncaught exception...
TypeError: curr.start.toISOString is not a function
    at Object.ical.objectHandlers.END (/home/pi/MagicMirror/modules/default/calendar/vendor/ical.js/node-ical.js:40:40)
    at Object.handleObject (/home/pi/MagicMirror/modules/default/calendar/vendor/ical.js/ical.js:400:41)
    at Object.parseICS (/home/pi/MagicMirror/modules/default/calendar/vendor/ical.js/ical.js:441:20)
    at Request._callback (/home/pi/MagicMirror/modules/default/calendar/vendor/ical.js/node-ical.js:11:24)
    at Request.self.callback (/home/pi/MagicMirror/node_modules/request/request.js:185:22)
    at Request.emit (events.js:182:13)
    at Request.<anonymous> (/home/pi/MagicMirror/node_modules/request/request.js:1157:10)
    at Request.emit (events.js:182:13)
    at Gunzip.<anonymous> (/home/pi/MagicMirror/node_modules/request/request.js:1079:12)
    at Object.onceWrapper (events.js:273:13)
```

sdetweil, 2019-06-09:
yep 
```
      rule += ' DTSTART:' + curr.start.toISOString().replace(/[-:]/g, '');
```
looks like the start of the event  hasn't been seen yet, only the end..
that file (node-ical.js) is in 
modules/default/calendar/vendor/ical.js

sdetweil, 2019-06-09:
i am just another user like you.. I didn't write any of this code, so don't really know what is supposed to be happening here...  sort of looks like if we didn't find a start of event, 
```
 if (rule.indexOf('DTSTART') === -1) {
```
fake one..
```
      if (curr.start.length === 8) {
        var comps = /^(\d{4})(\d{2})(\d{2})$/.exec(curr.start);
        if (comps) {
         curr.start = new Date (comps[1], comps[2] - 1, comps[3]);
        }
      }
```

Hg347, 2019-06-09:
Got it: curr.start was not a date but probably  a string when examining the ics file. It had nothing  to do with authentication.

Here is the fix of node-ical.js, lines 40 to 43.
Now the calendar is showing :)
```
if( typeof (curr.start) === "date") {
          rule += ' DTSTART:' + curr.start.toISOString().replace(/[-:]/g, '');
          rule = rule.replace(/\.[0-9]{3}/, '');
      }
```

sdetweil, 2019-06-09:
you should submit that as a pull request

Hg347, 2019-06-09:
How to do that? I think I do not have permissions for a PR. At least I can't push my bug branch of develop to origin.

sdetweil, 2019-06-09:
create a fork of the master see 
https://guides.github.com/activities/forking/

then git clone that fork to your mirror 

make your changes

commit them, then push that commit to your fork

then  create a pull request 

https://help.github.com/en/articles/creating-a-pull-request-from-a-fork

closed 2019-06-15 by Hg347