MagicMirrorOrg/MagicMirror #1697: How to use the 'node_helper', 'serialport'

nhpunch, 2019-06-10:
```
const NodeHelper = require('node_helper')
const Serialport = require("serialport");

module.exports = NodeHelper.create({

  start: function () {
  },

  socketNotificationReceived: function (notification, payload) {
    const self = this

    if (notification === 'CONFIG') {
      self.config = payload
      
    var serialport = new Serialport('/dev/ttyACM0', {   
		baudRate: 9600
		});

    var led = 0;

      serialport.open(function () {
        console.log('connect...');
        serialport.on('data', function(data) { // 아두이노로부터 전달된 데이터
          console.log('data received: ' + data);
        });

        setInterval( function() { // 2초마다 아두이노에게 문자열을 전송하는 예
          led = !led;
          serialport.write(led==true ? "1" : "0", function(err, result){
            if(err){
             console.log(err);
            }
          });
        }, 5000);
      });
    }
  }, 
})
```
this is node_helper
```
Module.register('MMM-Serial-Notifications', {
  defaults: {
  },

  getScripts: function() {
		return ["modules/MMM-Serial-Notifications/js/jquery.js"];
	},
	getStyles: function() {
		return ["mm-MMM-Serial-Notifications-style.css"];
	},
  start: function () {
    Log.info("Starting module: " + this.name);
    this.sendSocketNotification('CONFIG', this.config)
  },	
  socketNotificationReceived: function (notification, payload) {	
	this.sendNotification(notification, payload)

  },

	getDom: function() {
		var wrapper = document.createElement("div");
		var button = document.createElement("div");
		var text = document.createElement("span");
		var overlay = document.createElement("div");
		var hidden = true;

		overlay.className = "paint-it-black";
		button.className = "hide-toggle";
		button.appendChild(text);
		text.innerHTML = "ON";

		var led = 0;
                wrapper.appendChild(button);
		wrapper.appendChild(overlay);$(button).on("click", function(){
			if(hidden){
				$(overlay).fadeIn(1000);
				$(button).fadeTo(1000, 0.3);
				$(text).html('OFF');
				hidden = false;  
				serialport.write('0');
			}else{
				$(overlay).fadeOut(1000);
				$(button).fadeTo(1000, 1);  
				$(text).html('ON');
				hidden = true;
				serialport.write('1');
			}
		});
		return wrapper;
	}
});
```
this is module

I want to send ‘1’, ‘0’ to Arduino by pressing the button

Communication is possible using setInterval in node_helper

but, can not use serialport.write in a module

How do I use the serialport in the module?

Or, how do I create a button event in the node_helper?

I am a very beginner. Please let me know. 😥

sdetweil, 2019-06-10:
i posted to the forum for you, this is not a magicmirror problem

nhpunch, 2019-06-10:
@sdetweil 

serialport is working on node helper

I confirmed the communication with serial.write.

sdetweil, 2019-06-13:
we worked thru this in the fourm... all good?   please close issue if ok

closed 2019-06-14 by MichMich