MagicMirrorOrg/MagicMirror #1714: Travis builds failing on fork

sdetweil, 2019-06-29:
Please only submit reproducible issues. 

Worker information
hostname: 6ffd1187-2357-4409-87d7-129eea6de22b@1.production-2-worker-com-gce-p58r
version: v6.2.0 https://github.com/travis-ci/worker/tree/5e5476e01646095f48eec13196fdb3faf8f5cbf7
instance: travis-job-df415e6f-857c-46c9-ad9c-e383f4c066a4 travis-ci-sardonyx-xenial-1553530528-f909ac5 (via amqp)
startup: 9.157477235s
system_info
Build system information
Build language: node_js
Build group: stable
Build dist: xenial
Build id: 117396176
Job id: 212052222
Runtime kernel version: 4.15.0-1028-gcp
travis-build version: b8049365d
Build image provisioning date and time
Mon Mar 25 16:43:24 UTC 2019
Operating System Details
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.6 LTS
Release:	16.04
Codename:	xenial
Systemd Version
systemd 229
Cookbooks Version
42e42e4 https://github.com/travis-ci/travis-cookbooks/tree/42e42e4
git version
git version 2.21.0
bash version
GNU bash, version 4.3.48(1)-release (x86_64-pc-linux-gnu)
gcc version
gcc (Ubuntu 5.4.0-6ubuntu1~16.04.11) 5.4.0 20160609
docker version
Client:
 Version:           18.06.0-ce
 API version:       1.38
 Go version:        go1.10.3
 Git commit:        0ffa825
 Built:             Wed Jul 18 19:11:02 2018
 OS/Arch:           linux/amd64
 Experimental:      false
Server:
 Engine:
  Version:          18.06.0-ce
  API version:      1.38 (minimum version 1.12)
  Go version:       go1.10.3
  Git commit:       0ffa825
  Built:            Wed Jul 18 19:09:05 2018
  OS/Arch:          linux/amd64
  Experimental:     false
clang version
clang version 7.0.0 (tags/RELEASE_700/final)
jq version
jq-1.5
bats version
Bats 0.4.0
shellcheck version
0.6.0
shfmt version
v2.6.3
ccache version
3.2.4
cmake version
heroku version
heroku/7.22.7 linux-x64 node-v11.10.1
imagemagick version
Version: ImageMagick 6.8.9-9 Q16 x86_64 2018-09-28 http://www.imagemagick.org
md5deep version
4.4
mercurial version
version 4.8
mysql version
mysql  Ver 14.14 Distrib 5.7.25, for Linux (x86_64) using  EditLine wrapper
openssl version
OpenSSL 1.0.2g  1 Mar 2016
packer version
1.3.3
postgresql client version
psql (PostgreSQL) 10.7 (Ubuntu 10.7-1.pgdg16.04+1)
ragel version
Ragel State Machine Compiler version 6.8 Feb 2013
sudo version
1.8.16
gzip version
gzip 1.6
zip version
Zip 3.0
vim version
VIM - Vi IMproved 7.4 (2013 Aug 10, compiled Nov 24 2016 16:44:48)
iptables version
iptables v1.6.0
curl version
curl 7.47.0 (x86_64-pc-linux-gnu) libcurl/7.47.0 GnuTLS/3.4.10 zlib/1.2.8 libidn/1.32 librtmp/2.3
wget version
GNU Wget 1.17.1 built on linux-gnu.
rsync version
rsync  version 3.1.1  protocol version 31
gimme version
v1.5.3
nvm version
0.34.0
perlbrew version
/home/travis/perl5/perlbrew/bin/perlbrew  - App::perlbrew/0.86
phpenv version
rbenv 1.1.2
rvm version
rvm 1.29.7 (latest) by Michal Papis, Piotr Kuczynski, Wayne E. Seguin [https://rvm.io]
default ruby version
ruby 2.5.3p105 (2018-10-18 revision 65156) [x86_64-linux]
CouchDB version
couchdb 1.6.1
ElasticSearch version
5.5.0
mInstalled Firefox version
firefox 63.0.1
MongoDB version
MongoDB 4.0.7
PhantomJS version
2.1.1
Pre-installed PostgreSQL versions
9.4.21
9.5.16
9.6.12
Redis version
redis-server 5.0.4
Pre-installed Go versions
1.11.1
mvn version
Apache Maven 3.6.0 (97c98ec64a1fdfee7767ce5ffb20918da4f719f3; 2018-10-24T18:41:47Z)
gradle version
Gradle 4.10.2!
lein version
Leiningen 2.9.1 on Java 11.0.2 OpenJDK 64-Bit Server VM
Pre-installed Node.js versions
v10.15.3
v11.0.0
v4.9.1
v6.17.0
v8.12.0
v8.15.1
v8.9
phpenv versions
  system
  5.6
  5.6.40
  7.1
  7.1.27
  7.2
* 7.2.15 (set by /home/travis/.phpenv/version)
  hhvm
  hhvm-stable
composer --version
Composer version 1.8.4 2019-02-11 10:52:10
Pre-installed Ruby versions
ruby-2.3.8
ruby-2.4.5
ruby-2.5.3
docker_mtu
resolvconf
git.checkout
1.40s$ git clone --depth=50 --branch=newinstaller https://github.com/sdetweil/MagicMirror.git sdetweil/MagicMirror
Cloning into 'sdetweil/MagicMirror'...
remote: Enumerating objects: 8019, done.
remote: Counting objects: 100% (8019/8019), done.
remote: Compressing objects: 100% (3571/3571), done.
remote: Total 8019 (delta 4577), reused 7783 (delta 4370), pack-reused 0
Receiving objects: 100% (8019/8019), 7.74 MiB | 27.79 MiB/s, done.
Resolving deltas: 100% (4577/4577), done.
$ cd sdetweil/MagicMirror
$ git checkout -qf db62b7421a3179b807af0d34a8e61d564f4ba2da
nvm.install
3.29s$ nvm install 8
Downloading and installing node v8.16.0...
Downloading https://nodejs.org/dist/v8.16.0/node-v8.16.0-linux-x64.tar.xz...
Computing checksum with sha256sum
Checksums matched!
Now using node v8.16.0 (npm v6.4.1)
cache.1
Setting up build cache
$ export CASHER_DIR=${TRAVIS_HOME}/.casher
0.29s$ Installing caching utilities
0.00s2.25sattempting to download cache archive
fetching newinstaller/cache-linux-xenial-e3b0c44298fc1c149afbf4c8996fb92427ae41e4649b934ca495991b7852b855--node-8.tgz
fetching newinstaller/cache--node-8.tgz
fetching master/cache-linux-xenial-e3b0c44298fc1c149afbf4c8996fb92427ae41e4649b934ca495991b7852b855--node-8.tgz
fetching master/cache--node-8.tgz
could not download cache
0.00s1.23sadding /home/travis/build/sdetweil/MagicMirror/node_modules to cache
creating directory /home/travis/build/sdetweil/MagicMirror/node_modules
$ node --version
v8.16.0
$ npm --version
6.4.1
$ nvm --version
0.34.0
install.npm
23.49s$ npm ci 
npm WARN prepare removing existing node_modules/ before installation
> electron@3.0.13 postinstall /home/travis/build/sdetweil/MagicMirror/node_modules/electron
> node install.js
> electron-chromedriver@1.8.0 install /home/travis/build/sdetweil/MagicMirror/node_modules/electron-chromedriver
> node ./download-chromedriver.js
successfully dowloaded and extracted!
> magicmirror@2.7.1 install /home/travis/build/sdetweil/MagicMirror
> cd vendor && npm install
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.4 (node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.4: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})
added 81 packages from 56 contributors and audited 219 packages in 3.091s
found 5 vulnerabilities (4 low, 1 high)
  run `npm audit fix` to fix them, or `npm audit` for details
> magicmirror@2.7.1 postinstall /home/travis/build/sdetweil/MagicMirror
> sh installers/postinstall/postinstall.sh && npm run install-fonts
MagicMirror installation successful!
> magicmirror@2.7.1 install-fonts /home/travis/build/sdetweil/MagicMirror
> cd fonts && npm install
added 1 package from 1 contributor and audited 1 package in 0.956s
found 0 vulnerabilities
added 926 packages in 23.128s
before_script.1
0.92s$ yarn danger ci
yarn run v1.15.2
$ /home/travis/build/sdetweil/MagicMirror/node_modules/.bin/danger ci
You may have updated from Danger 2.x -> 3.x without updating from `danger` to `danger ci`.
Skipping Danger due to this run not executing on a PR.
before_script.2
5.04s$ npm install grunt-cli -g
/home/travis/.nvm/versions/node/v8.16.0/bin/grunt -> /home/travis/.nvm/versions/node/v8.16.0/lib/node_modules/grunt-cli/bin/grunt
+ grunt-cli@1.3.2
added 150 packages from 121 contributors in 4.606s
before_script.3
0.00s$ export DISPLAY=:99.0
0.00s$ sh -e /etc/init.d/xvfb start
sh: 0: Can't open /etc/init.d/xvfb
The command "sh -e /etc/init.d/xvfb start" failed and exited with 127 during .
Your build has been stopped.

roramirez, 2019-08-11:
Hi @sdetweil this happend because the image using in default was `xenial` with PR #1748 fixed this issue. Please update your upstream branch from `develop` and should not failed.

MichMich, 2019-10-04:
Solved

closed 2019-10-04 by MichMich