MagicMirrorOrg/MagicMirror #1718: Calendar: All items scheduled for one day late

Kugelfang666, 2019-07-08:
lease only submit re
When submitting a new issue, please supply the following information:

**Platform**: Raspberry Pi 2/3

**MagicMirror Version**: latest version pulled from git 2019-07-07

**Description**: 

The calendar plugin shows items on wrong date. I'm connected to two iCloud calendars. An Item which is scheduled for tomorrow appears to be due for the day after tomorrow. All other plugins like the clock are showing correct date + time

**Additional Notes**: this problem occurred once I updated the installation. Previously I was running a version dating back to April. Problem occurred without touching the config.js

Kugelfang666, 2019-07-11:
For the record: I rolled back the default calendar plugin to the previous release (April) an the problem is solved. There seems to be something wrong with the latest release

Kugelfang666, 2019-07-18:
Unfortunately rolling back the calendar module created a new problem. Now the calendar gets stuck at one day. Hence after a fresh boot it correctly shows that a item is due „tomorrow“ but it does not change to „today“ at midnight. In fact the items displayed are stuck at the day of booting

MichMich, 2019-07-21:
We did; t get any other reports of this issue. Have you checked if the timezone is correct?

Kugelfang666, 2019-07-21:
Hi, mhm actually I did, both in MM as well as via SSH in raspy-config...

With regards to the initial problem (one day lag): this was solved by rolling back to the previous version of the calendar module.

The second problem was actually my mistake: I set the module to a rather long fetch interval of twice a day to prevent my Synology which is hosting the calendars to wake up unnecessarily often. Could it be that the fetch interval also governs how often the time indications (tomorrow, today...) are being updated? After decreasing the setting the problem went away. 

Maybe it would be worthwhile to introduce two separate values, one for how often the calendar is fetched from the server and a second for how often the interface is being updated.

stefsims, 2019-08-13:
I also have this issue. All day events "today" are show as "tomorrow".
This is only for recurring events. I made an entry only for today and it showed "today".
Using Google Calendar
My config:
	{
		colored: true,
		maximumEntries: 5,
		maximumNumberOfDays: 3,
		timeFormat: 'absolute',
		showEnd: false,
		fullDayEventDateFormat: 'dddd',
		dateFormat: 'HH:mm ddd',
		dateEndFormat: 'HH:mm',
		urgency: 0,
		getRelative: 8,

magnuskonze, 2019-08-31:
I also recognised this problem.

I'm new to the MagicMirror-Project (tried it this week for the first time), but i already played around with the code:
In calendarfetcher.js line 187 (var dates = rule.between(past, future, true, limitFunction);) all dates are off by 2 hours.
So line 228 (startDate = moment(date);) creats a wrong startDate. When i remove this line everything seems to work fine. I still don't understand why the date has to be recalculated here.

I noticed, that my google calendar has timezone GMT +2 (Berlin), while my raspberry has Europe/Berlin. Should be the same, but it feels like a timezone problem.

Perhaps this helps?

magnuskonze, 2019-09-08:
My "fix" (removing line 228) doesn't work anymore. I can't reproduce the scenario anymore...

Anyway, the problem only occures with recurring fullday events. They are shown one day later.
Recurring events with starting time respectively non-recurring fullday events are working fine.

Btw. those  calendar rules seem to use UTC, correct? This would explain, while the dates in line 187 are 2 hours off in my case.

magnuskonze, 2019-09-23:
I switched back to version 2.7.1 and now recurring fullday events seem to work for me.

Kugelfang666, 2019-11-03:
I just ran an full update on my MM and the issue reappeared. After rolling just the calendar module back tu 2.7.1. everything is working fine again.

ykuzmycz, 2019-11-14:
@Kugelfang666 how do you roll back just the calendar module?

Kugelfang666, 2019-11-14:
I just downloaded the folder from an previous release from git

ykuzmycz, 2019-11-14:
I can confirm rolling back just the calendar module to 2.7.1 corrects my reoccurring events (both daily and hourly).

#1803 
#1798

oemel09, 2020-01-02:
I noticed, that this depends on the way the event saves the recurrence rule.
I had my New Years Eve event saved with this rrule:
```
rrule: RRule {
    _cache: Cache { all: false, before: [], after: [], between: [] },
    origOptions: {
      dtstart: 2000-12-30T23:00:00.000Z,
      freq: 0,
      bymonth: 12,
      bymonthday: 31,
      tzid: undefined
    },
    ...
},
```

New Years Day event was saved with this rrule:
```
rrule: RRule {
    _cache: Cache { all: false, before: [], after: [], between: [] },
    origOptions: {
      dtstart: 1999-12-31T23:00:00.000Z,
      freq: 0,
      interval: 1,
      tzid: undefined
    },
    ....
},
```

In my calendar app they were both displayed correctly.

However after calling the `between` function the returned date for both events was `2019-12-31T23:00:00.000Z` resulting in displaying New Years Eve and New Years Day happening at the same day.

I assume 23:00:00 is because those events are full day events and are saved in UTC as my time zone currently is UTC+1.

closed 2020-03-10 by stale[bot]