MagicMirrorOrg/MagicMirror #1726: Latest MagicMirror out of memory on raspi 3b

paaland, 2019-07-13:
**Platform**: Raspberry Pi 3
**Node Version**: v10.16.0

**MagicMirror Version**: JavaScript version (v2.8.0)

**Description**: After the last update MM will run for about a day before the screen just goes black. dmesg show's electron was killed due to out-of-memory issue:

```
[90137.171173] Out of memory: Kill process 19644 (electron) score 322 or sacrifice child
[90137.171235] Killed process 19644 (electron) total-vm:314276kB, anon-rss:18856kB, file-rss:0kB, shmem-rss:0kB
[90137.182175] oom_reaper: reaped process 19644 (electron), now anon-rss:0kB, file-rss:0kB, shmem-rss:0kB
```

**Steps to Reproduce**: I've got Clock, MMM-YrNow, MMM-Anti-Screen-BurnIn and netatmo modules.

**Expected Results**: Before the latest release MM ran for ages without issues. This latest version runs out of memory after about a day.

**Configuration**: 
```
var config = {
        address: "0.0.0.0", // Address to listen on, can be:
        port: 8080,
        ipWhitelist: [], // Set [] to allow all IP addresses

        language: "nb",
        timeFormat: 24,
        units: "metric",

        modules: [
                {
                        module: "alert",
                },
                {
                        module: "updatenotification",
                        position: "top_bar"
                },
                {
                        module: "clock",
                        position: "top_left"
                },
                {
                                module: "MMM-Anti-Screen-Burn-In",
                                position: "middle"
                },
                {
                        module: 'MMM-YrNow',
                        position: 'bottom_left',
                        config: {
                                locationId: '1-30849',
                                showWeatherForecast: true
                        }
                },
                {
                        module: 'netatmo',
                        position: 'top_right', // the location where the module should be displayed
                        config: {
                                clientId: 'xxxxxxxxxxx',
                                clientSecret: 'xxxxxxxxxxxxx',
                                refreshToken: 'xxxxxxxxxxxxxxxx',
                                moduleOrder: ['Ute', 'Stue', 'Regn'],
                                dataOrder: ['Temperature', 'Humidity', 'Pressure', 'Rain'],
                                design: 'bubbles',
                                hideLoadTimer: false,
                                showBattery: false,
                                showTrend: false,
                                showRadio: false,
                                showWifi: false,
                                showLastMessage: false,
                                showCO2: false,
                                horizontal: false
                        }
                }
        ]
};

/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== "undefined") { module.exports = config; }
```

MichMich, 2019-07-21:
Could you try running it without 3rd party modules?

sdetweil, 2019-09-20:
also, enlarge the swap space on the pi.. I had failures at the default 100meg.. made it 1 gig and all is well. 
see
https://wpitchoune.net/tricks/raspberry_pi3_increase_swap_size.html
and reboot

MichMich, 2019-10-04:
If this problem still occurs, feel free to reopen this issue.

closed 2019-10-04 by MichMich