MagicMirrorOrg/MagicMirror #1731: install electron fail

dfybc, 2019-07-17:
Please only submit reproducible issues. 

If you're not sure if it's a real bug or if it's just you, please open a topic on the forum: [https://forum.magicmirror.builders/category/15/bug-hunt](https://forum.magicmirror.builders/category/15/bug-hunt)
Problems installing or configuring your MagicMirror? Check out: [https://forum.magicmirror.builders/category/10/troubleshooting](https://forum.magicmirror.builders/category/10/troubleshooting)

When submitting a new issue, please supply the following information:

**Platform**: Place your platform here... give us your web browser/Electron version *and* your hardware (Raspberry Pi 2/3, Windows, Mac, Linux, System V UNIX).

**Node Version**: Make sure it's version 0.12.13 or later.

**MagicMirror Version**: Now that the versions have split, tell us if you are using the PHP version (v1) or the newer JavaScript version (v2).

**Description**: Provide a detailed description about the issue and include specific details to help us understand the problem. Adding screenshots will help describing the problem.

**Steps to Reproduce**: List the step by step process to reproduce the issue.

**Expected Results**: Describe what you expected to see.

**Actual Results**: Describe what you actually saw.

**Configuration**: What does the used config.js file look like? Don't forget to remove any sensitive information!

**Additional Notes**: Provide any other relevant notes not previously mentioned. This is optional.

dfybc, 2019-07-17:
RPi 3B+
Node version 10.16.0
when i do  "npm install", the following error happened? 
can anyone give some advice?

pi@raspberrypi:~/MagicMirror $ sudo npm install
npm WARN deprecated time-grunt@2.0.0: Deprecated because Grunt is practically unmaintained. Move on to something better. This package will continue to work with Grunt v1, but it will not receive any updates.

> electron-chromedriver@1.8.0 install /home/pi/MagicMirror/node_modules/electron-chromedriver
> node ./download-chromedriver.js

/home/pi/MagicMirror/node_modules/electron-chromedriver/download-chromedriver.js:30
  if (err != null) throw err
                   ^

Error: EACCES: permission denied, mkdir '/home/pi/MagicMirror/node_modules/electron-chromedriver/.electron'
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! electron-chromedriver@1.8.0 install: `node ./download-chromedriver.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the electron-chromedriver@1.8.0 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-07-17T08_47_52_650Z-debug.log
pi@raspberrypi:~/MagicMirror $

sdetweil, 2019-07-19:
>sudo npm install

do NOT use sudo.. all kinds of trouble later

to fix, from /home/pi  folder, do 
```
sudo chown -R pi MagicMirror 
```
then redo the npm install from the MagicMirror folder

closed 2019-07-21 by MichMich