MagicMirrorOrg/MagicMirror #1732: My Magic Mirror Was working very good until I did update it

SamerAl-khateeb, 2019-07-18:
I really appreciate any help here.
I have installed MagicMirror and it was woking great. I saw there are two updates so I did update it and not my MagicMirror does not work. It start but I see a black screen! Any Idea why?

P.s. I am using Raspberry Pi 3 B+ and Raspbian Stretch!

Also, I am getting many errors which I am showing below!

pi@raspberrypi:~ $ cd MagicMirror
pi@raspberrypi:~/MagicMirror $ git pull && npm install
npm WARN checkPermissions Missing write access to /home/pi/MagicMirror/node_modules/snowboy/node_modules/assert-plus
npm WARN checkPermissions Missing write access to /home/pi/MagicMirror/node_modules/snowboy/node_modules/aws-sign2
npm WARN checkPermissions Missing write access to /home/pi/MagicMirror/node_modules/snowboy/node_modules/camelcase
npm WARN checkPermissions Missing write access to /home/pi/MagicMirror/node_modules/snowboy/node_modules/cli-spinners
npm WARN checkPermissions Missing write access to /home/pi/MagicMirror/node_modules/snowboy/node_modules/cliui/node_modules/ansi-regex
npm WARN checkPermissions Missing write access to /home/pi/MagicMirror/node_modules/snowboy/node_modules/cliui/node_modules/is-fullwidth-code-point
npm WARN checkPermissions Missing write access to /home/pi/MagicMirror/node_modules/snowboy/node_modules/cliui/node_modules/strip-ansi
npm WARN checkPermissions Missing write access to /home/pi/MagicMirror/node_modules/snowboy/node_modules/cliui/node_modules/string-width
npm WARN checkPermissions Missing write access to /home/pi/MagicMirror/node_modules/snowboy/node_modules/clone
npm WARN checkPermissions Missing write access to /home/pi/MagicMirror/node_modules/snowboy/node_modules/color-name
npm WARN checkPermissions Missing write access to /home/pi/MagicMirror/node_modules/snowboy/node_modules/color-convert
npm WARN checkPermissions Missing write access to /home/pi/MagicMirror/node_modules/snowboy/node_modules/ansi-styles
npm WARN checkPermissions Missing write access to /home/pi/MagicMirror/node_modules/snowboy/node_modules/colors
npm WARN checkPermissions Missing write access to /home/pi/MagicMirror/node_modules/snowboy/node_modules/decamelize
npm WARN checkPermissions Missing write access to /home/pi/MagicMirror/node_modules/snowboy/node_modules/defaults
npm WARN checkPermissions Missing write access to /home/pi/MagicMirror/node_modules/snowboy/node_modules/escape-string-regexp
npm WARN checkPermissions Missing write access to /home/pi/MagicMirror/node_modules/snowboy/node_modules/fast-deep-equal
npm WARN checkPermissions Missing write access to /home/pi/MagicMirror/node_modules/snowboy/node_modules/fast-json-stable-stringify
npm WARN checkPermissions Missing write access to /home/pi/MagicMirror/node_modules/snowboy/node_modules/get-caller-file
npm WARN checkPermissions Missing write access to /home/pi/MagicMirror/node_modules/snowboy/node_modules/har-schema
npm WARN checkPermissions Missing write access to /home/pi/MagicMirror/node_modules/snowboy/node_modules/has-flag
npm WARN checkPermissions Missing write access to /home/pi/MagicMirror/node_modules/snowboy/node_modules/invert-kv
npm WARN checkPermissions Missing write access to /home/pi/MagicMirror/node_modules/snowboy/node_modules/is-stream
npm WARN checkPermissions Missing write access to /home/pi/MagicMirror/node_modules/snowboy/node_modules/isexe
npm WARN checkPermissions Missing write access to /home/pi/MagicMirror/node_modules/snowboy/node_modules/json-schema-traverse
npm WARN checkPermissions Missing write access to /home/pi/MagicMirror/node_modules/snowboy/node_modules/jsonfile
npm WARN checkPermissions Missing write access to /home/pi/MagicMirror/node_modules/snowboy/node_modules/lcid
npm WARN checkPermissions Missing write access to /home/pi/MagicMirror/node_modules/snowboy/node_modules/lodash.assign
npm WARN checkPermissions Missing write access to /home/pi/MagicMirror/node_modules/snowboy/node_modules/mem/node_modules/mimic-fn
npm WARN checkPermissions Missing write access to /home/pi/MagicMirror/node_modules/snowboy/node_modules/form-data
npm WARN checkPermissions Missing write access to /home/pi/MagicMirror/node_modules/snowboy/node_modules/mimic-fn
npm WARN checkPermissions Missing write access to /home/pi/MagicMirror/node_modules/snowboy/node_modules/ms
npm WARN checkPermissions Missing write access to /home/pi/MagicMirror/node_modules/snowboy/node_modules/debug
npm WARN checkPermissions Missing write access to /home/pi/MagicMirror/node_modules/snowboy/node_modules/nice-try
npm WARN checkPermissions Missing write access to /home/pi/MagicMirror/node_modules/snowboy/node_modules/node-gyp/node_modules/semver
npm WARN checkPermissions Missing write access to /home/pi/MagicMirror/node_modules/snowboy/node_modules/nopt
npm WARN checkPermissions Missing write access to /home/pi/MagicMirror/node_modules/snowboy/node_modules/oauth-sign
npm WARN checkPermissions Missing write access to /home/pi/MagicMirror/node_modules/snowboy/node_modules/onetime
npm WARN checkPermissions Missing write access to /home/pi/MagicMirror/node_modules/snowboy/node_modules/ora/node_modules/ansi-regex
npm WARN checkPermissions Missing write access to /home/pi/MagicMirror/node_modules/snowboy/node_modules/ora/node_modules/strip-ansi
npm WARN checkPermissions Missing write access to /home/pi/MagicMirror/node_modules/snowboy/node_modules/p-defer
npm WARN checkPermissions Missing write access to /home/pi/MagicMirror/node_modules/snowboy/node_modules/map-age-cleaner
npm WARN checkPermissions Missing write access to /home/pi/MagicMirror/node_modules/snowboy/node_modules/p-finally
npm WARN checkPermissions Missing write access to /home/pi/MagicMirror/node_modules/snowboy/node_modules/p-is-promise
npm WARN checkPermissions Missing write access to /home/pi/MagicMirror/node_modules/snowboy/node_modules/mem
npm WARN checkPermissions Missing write access to /home/pi/MagicMirror/node_modules/snowboy/node_modules/p-try
npm WARN checkPermissions Missing write access to /home/pi/MagicMirror/node_modules/snowboy/node_modules/p-limit
npm WARN checkPermissions Missing write access to /home/pi/MagicMirror/node_modules/snowboy/node_modules/p-locate
npm WARN checkPermissions Missing write access to /home/pi/MagicMirror/node_modules/snowboy/node_modules/path-exists
npm WARN checkPermissions Missing write access to /home/pi/MagicMirror/node_modules/snowboy/node_modules/locate-path
npm WARN checkPermissions Missing write access to /home/pi/MagicMirror/node_modules/snowboy/node_modules/find-up
npm WARN checkPermissions Missing write access to /home/pi/MagicMirror/node_modules/snowboy/node_modules/path-key
npm WARN checkPermissions Missing write access to /home/pi/MagicMirror/node_modules/snowboy/node_modules/npm-run-path
npm WARN checkPermissions Missing write access to /home/pi/MagicMirror/node_modules/snowboy/node_modules/performance-now
npm WARN checkPermissions Missing write access to /home/pi/MagicMirror/node_modules/snowboy/node_modules/psl
npm WARN checkPermissions Missing write access to /home/pi/MagicMirror/node_modules/snowboy/node_modules/punycode
npm WARN checkPermissions Missing write access to /home/pi/MagicMirror/node_modules/snowboy/node_modules/qs
npm WARN checkPermissions Missing write access to /home/pi/MagicMirror/node_modules/snowboy/node_modules/require-directory
npm WARN checkPermissions Missing write access to /home/pi/MagicMirror/node_modules/snowboy/node_modules/require-main-filename
npm WARN checkPermissions Missing write access to /home/pi/MagicMirror/node_modules/snowboy/node_modules/node-abi
npm WARN checkPermissions Missing write access to /home/pi/MagicMirror/node_modules/snowboy/node_modules/shebang-regex
[... 44 lines cut ...]
npm WARN checkPermissions Missing write access to /home/pi/MagicMirror/node_modules/snowboy/node_modules/yargs/node_modules
npm WARN checkPermissions Missing write access to /home/pi/MagicMirror/node_modules/snowboy/node_modules/spawn-rx/node_modules
npm ERR! path /home/pi/MagicMirror/node_modules/snowboy/node_modules/assert-plus
npm ERR! code EACCES
npm ERR! errno -13
npm ERR! syscall access
npm ERR! Error: EACCES: permission denied, access '/home/pi/MagicMirror/node_modules/snowboy/node_modules/assert-plus'
npm ERR!  { [Error: EACCES: permission denied, access '/home/pi/MagicMirror/node_modules/snowboy/node_modules/assert-plus']
npm ERR!   stack:
npm ERR!    'Error: EACCES: permission denied, access \'/home/pi/MagicMirror/node_modules/snowboy/node_modules/assert-plus\'',
npm ERR!   errno: -13,
[... cut ...]
npm ERR! The operation was rejected by your operating system.
npm ERR! It is likely you do not have the permissions to access this file as the current user
npm ERR! 
npm ERR! If you believe this might be a permissions issue, please double-check the
npm ERR! permissions of the file and its containing directories, or try running
npm ERR! the command again as root/Administrator (though this is not recommended).

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2019-07-18T04_31_49_325Z-debug.log

sdetweil, 2019-07-19:
looks like you did sudo npm install on the first install.. and not this time.

to fix, from the MagicMirror folder  do 
```
sudo chown -R pi node_modules
```
then redo the npm install (from the MagicMirror folder)

SamerAl-khateeb, 2019-07-20:
@sdetweil thank you so much for the help. I think the update went through but MagicMirror still show black screen! I do not know why though. Probably, I have to reinstall everything from the begining. Here is the steps I did with no errors:

pi@raspberrypi:~/MagicMirror $ sudo chown -R pi node_modules
pi@raspberrypi:~/MagicMirror $ sudo git pull && npm install
Already up-to-date.

> core-js@2.6.9 postinstall /home/pi/MagicMirror/node_modules/core-js
> node scripts/postinstall || echo "ignore"

Thank you for using core-js ( https://github.com/zloirock/core-js ) for polyfilling JavaScript standard library!

The project needs your help! Please consider supporting of core-js on Open Collective or Patreon: 
> https://opencollective.com/core-js 
> https://www.patreon.com/zloirock 

Also, the author of core-js ( https://github.com/zloirock ) is looking for a good job -)

> electron@3.1.10 postinstall /home/pi/MagicMirror/node_modules/electron
> node install.js

Downloading tmp-4881-1-SHASUMS256.txt-3.1.10
[============================================>] 100.0% of 4.79 kB (4.79 kB/s)

> magicmirror@2.8.0 install /home/pi/MagicMirror
> cd vendor && npm install

npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.4 (node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.4: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})

added 1 package from 1 contributor and audited 220 packages in 8.266s
found 5 vulnerabilities (4 low, 1 high)
  run `npm audit fix` to fix them, or `npm audit` for details

> magicmirror@2.8.0 postinstall /home/pi/MagicMirror
> sh installers/postinstall/postinstall.sh && npm run install-fonts

MagicMirror installation successful!

> magicmirror@2.8.0 install-fonts /home/pi/MagicMirror
> cd fonts && npm install

audited 1 package in 1.01s
found 0 vulnerabilities

added 147 packages from 161 contributors, removed 371 packages, updated 209 packages, moved 5 packages and audited 3574 packages in 260.199s
found 18 high severity vulnerabilities
  run `npm audit fix` to fix them, or `npm audit` for details
pi@raspberrypi:~/MagicMirror $ npm audit fix
updated 1 package in 30.968s
fixed 18 of 18 vulnerabilities in 3574 scanned packages
pi@raspberrypi:~/MagicMirror $

SamerAl-khateeb, 2019-07-20:
@sdetweil 
Update: I figured the screen was black because of one of the modules which I will check why its not working later. But now it works. Thank you very much for the help.

closed 2019-07-20 by SamerAl-khateeb

sdetweil, 2019-07-20:
You might also need to do 
npm install
For each module with a package.json file