MagicMirrorOrg/MagicMirror #1734: Dependencies will not install and autostart

Garretto123, 2019-07-24:
Please only submit reproducible issues. 

If you're not sure if it's a real bug or if it's just you, please open a topic on the forum: [https://forum.magicmirror.builders/category/15/bug-hunt](https://forum.magicmirror.builders/category/15/bug-hunt)
Problems installing or configuring your MagicMirror? Check out: [https://forum.magicmirror.builders/category/10/troubleshooting](https://forum.magicmirror.builders/category/10/troubleshooting)

When submitting a new issue, please supply the following information:

**Platform**: Place your platform here... give us your web browser/Electron version *and* your hardware (Raspberry Pi 2/3, Windows, Mac, Linux, System V UNIX).
Raspberry Pi3 B+
**Node Version**: Make sure it's version 0.12.13 or later.
The latest
**MagicMirror Version**: Now that the versions have split, tell us if you are using the PHP version (v1) or the newer JavaScript version (v2).
Not sure
**Description**: Provide a detailed description about the issue and include specific details to help us understand the problem. Adding screenshots will help describing the problem.
I am having trouble downloading the magic mirror software. So on June 25th, I downloaded the MM software successfully. After copying the one line of code, I went through all the steps and it all worked. After that day, I lost my SD card with the MM software on so I had to restart it. I didn't try to re-download it until 2 days ago and nothing is working. When I copy the code into the terminal, the prompts load up. Then it says I don't have the latest version of node. And it also says dependencies have not installed. So to fix that, I downloaded the newest version of node.js and the software kind of worked. The only thing that didn't work was auto-start. I tried everything to try and make it auto-start but still doesn't work. I am just confused because before the MM update, it was all working and after, it isn't even though I am doing the exact same process. Someone please help. I don't know what to do anymore.
**Steps to Reproduce**: List the step by step process to reproduce the issue.

**Expected Results**: Describe what you expected to see.
The mirror autostarting and not saying that dependencies not installed.
**Actual Results**: Describe what you actually saw.

**Configuration**: What does the used config.js file look like? Don't forget to remove any sensitive information!
I am not sure what the config looks like although I do believe that it is not that problem.
**Additional Notes**: Provide any other relevant notes not previously mentioned. This is optional.

sdetweil, 2019-07-24:
sounds like npm is downlevel too.. 
npm -v 
should show V6.x.x
if not then do 
```
sudo npm i -g npm
```
then erase the node_modules folder 
```
rm -rf node_modules
and install dependencies again
npm install
```

Garretto123, 2019-07-24:
Thank you. I was able to get the mirror working but the autostart didn't work. I powered it off and when I turned on the pi, the "MM booting up" page was on my screen for a second then it went away and went to the regular raspbian background. Any suggestions?

sdetweil, 2019-07-24:
sounds like the script failed..

do
```
pm2 status
```
then pm2 stop 0

need to know where the script is. installers/mm.sh
or MagicMirror/mm.sh

edit either one and change cd ~/maggicMirror  to 
cd /home/pi/MagicMirror

then try to reboot again

Garretto123, 2019-07-25:
Sorry. I’m a newbie at this stuff. The script is in the installers. /home/pi/MagicMirror/installers

When I tried clicking on mm.sh, the screen just turned black. It was the same as the MM screen just with nothing on it. 
When you mean edit do I just change the name. And what’s the exact file I am editing. I’m a little confused here

sdetweil, 2019-07-25:
Sounds like a problem w mirror black screen

So, let's back up

From the MagicMirror folder do
```
npm start dev &
```

Let's get that fixed
Select the console tab in the white side of the screen, and scroll up to see any errors

Garretto123, 2019-07-25:
Alright so the mirror interface correctly loaded. I could see the time, calendar, and so on. I checked to see the errors and the only thing that came up was an error that said my api for weather was wrong which makes sense because I have't changed the default one yet.
currentweather: Incorrect APPID.
The only issue is the autostart I think

sdetweil, 2019-07-25:
Hmmm,  mm.sh should do the same thing.

Edit the mm.sh and add dev
Onto the last line
So it ends
npm start dev.

Then click on that again

Garretto123, 2019-07-25:
Alright. I changed the mm.sh to
npm start dev.
and it runs the mm program just like it should. I then rebooted the pi to see if it would work on startup and it have me the same issue which was showing the booting up screen then not actually working

closed 2019-07-26 by MichMich