MagicMirrorOrg/MagicMirror #1735: Magic Mirror

9170895137, 2019-07-26:
I'm new to this. I'm in the process of building a magic mirror and I bought a LoveRPi Raspberry Pi 3 B+ 8GB Quick Start Kit with Raspbian Desktop, Raspberry Pi Mini Camera Video Module 5 Megapixels 1080p Sensor OV5647 Webcam for Raspberry Pi Model A/B/B+, Pi 2 and Raspberry Pi 3, 3b+ and 
Raspberry Pi RASPBERRYPI-SENSEHAT Sense HAT with Orientation, Pressure, Humidity and Temperature Sensors. The issues I'm having is loading the software for the magic mirror. I have tried numerous times and everything starts to load but than toward the end is fails to load the last batches of files. I did get the screen to go vertical and I did a software update under preferences and everything is up to date but I still can not get the magic mirror file to work. I could use some serious help. Or do I need to add more software?

NamanSharma2703, 2019-07-26:
Can you specify your exact problem?? Also is that your number 9170895137? Are u from India?

9170895137, 2019-07-26:
no I'm in reno nv

9170895137, 2019-07-26:
raspberry.sh | bash how do you get that symbol between sh and bash?

closed 2019-07-26 by MichMich