MagicMirrorOrg/MagicMirror #1736: Video Chat

9170895137, 2019-07-26:
Is there a module to allow video chat? I have alexa and a alexa spot and was wondering how I can tie them into the magic mirror. I got it finally running but how do I change the layout and add modules?

MichMich, 2019-07-26:
http://modules.magicmirror.builders

MichMich, 2019-07-26:
Please refer to http://forum.magicmirror.builders for support and troubleshooting.

I'm closing this issue for now. If you feel this issue should remain open, please let me know.

closed 2019-07-26 by MichMich

9170895137, 2019-07-26:
I found out how to get tot he terminal screen but about 10 seconds later the magic mirror starts back up

sdetweil, 2019-07-26:
thats what the pm2 software does.. it keeps the mirror running... alt-spacebar, n will minimize the mirror app (just like windows).. so you can do pm2 stop 0 

u need to read the documentation, and go to the forum as Mich has mentioned.

9170895137, 2019-07-26:
Im new at this 

From: sam detweiler
Sent: Friday, July 26, 2019 4:16 AM
To: MichMich/MagicMirror
Cc: 9170895137; Author
Subject: Re: [MichMich/MagicMirror] Video Chat (#1736)

thats what the pm2 software does.. it keeps the mirror running... alt-spacebar, n will minimize the mirror app (just like windows).. so you can do pm2 stop 0
u need to read the documentation, and go to the forum as Mich has mentioned.

sdetweil, 2019-07-26:
we can tell...
go here  https://forum.magicmirror.builders/category/10/troubleshooting
ask questions.. Mykle1 has some great guides for beginners..